OneOfEleven/uv-k5-firmware-custom #84: The possibility of setting a Second PTT key

CaptainRi, 2023-10-01:
First, your firmware is very interesting and practical. 

Can you set the custom 1 key as the second PTT key? The original PTT key controls the frequency above, and the custom 1 key controls the frequency below, so there is no need to frequently switch frequencies

OneOfEleven, 2023-10-01:
People use that key for various other things, I use myself use it for monitor on/off.

bricky149, 2023-10-01:
I've received a request to do this as well. I was thinking of using both SIDE1 and SIDE2 to TX on top or bottom VFO respectively and using KEY_PTT as the monitor. Last time I tried it was a tad messy, maybe there's a better way to do it.

egzumer, 2023-10-02:
I would love that as an option too, it doesn't have to be fixed. Buttons could stay configurable, but PTT2 could replace e.g. Alarm function.

OneOfEleven, 2023-10-02:
Am lagging behind bug fixes at the moment, need to catch up with them before adding more options I think, need to keep on top of them.

bricky149, 2023-10-06:
An update on this.

#134 is something that may affect my progress on this feature request. I believe it's possible to build this on the cross-band functionality that exists, but I'm understanding it's a bit of a mess currently. For now, I can only apologise for the delay.

OneOfEleven, 2023-10-06:
I've added 4 new menu items tonight, they allow the user to configure the 2 side buttons without needing the PC.

An option could be added to set a chosen side button to be your 2nd PTT.

bricky149, 2023-10-06:
> I've added 4 new menu items tonight, they allow the user to configure the 2 side buttons without needing the PC.
> 
> An option could be added to set a chosen side button to be your 2nd PTT.

Your crystal ball's on point this evening. I'll delve into it tomorrow. Sleep well.