OneOfEleven/uv-k5-firmware-custom #86: CTCSS ≥ 210,7Hz does not TX/RX properly.

ea7hjb, 2023-10-01:
Hello, I noted that analog subtones between 210,7 Hz and 250,3Hz does not work. No proper TX nor RX. Checked against an SDR and other quality HTs (Yaesu, Icom, etc).

Tested with ENABLE_CTCSS_TAIL_PHASE_SHIFT := 0 and 1 (just in case), an got same result.

In theory, the calibration data of my Quansheng is untouched, so I don't know it is my unit failing or a firmware bug, since I don't have other Quansheng, but this behaviour doesn't occur with original untouched firmware.

[image]

OneOfEleven, 2023-10-01:
oh, sorry about that, I'll look into it tomorrow, very late here now.

OneOfEleven, 2023-10-02:
Found the problem

Thank you for letting me know.

closed 2023-10-02 by OneOfEleven