OneOfEleven/uv-k5-firmware-custom #100: Wrong behaviuor of squelch, commit bf2564b

pawol, 2023-10-02:
Wrong behaviuor of squelch on  commit bf2564b.
Compiled with standard squelch. Squelch set quite high (value 5) in a silent environment
Green led blinks randomly very often. 
Previoulsly vale5 ensured complete muting

bricky149, 2023-10-02:
Are you using the squelch_lower option? I've found it can open squelch too keenly, perhaps due to the noise threshold.

pawol, 2023-10-02:
like I wrote - standard squelch

bricky149, 2023-10-02:
> like I wrote - standard squelch. I suspect auto bandwidth for small signals/ sensitivity mechanism you applied last time

My apologies for misunderstanding.

The issue you describe was one I experienced before that change. For me it was down to squelch, I'm not sure what your case might be.

OneOfEleven, 2023-10-03:
I have made the squelch more sensitive, not just in level but also in time, ie it opens/closes faster than before - to let me increase channel scan speed.

pawol, 2023-10-03:
green led blinks and backlight lits also. So, it is quite bad feature

OneOfEleven, 2023-10-03:
OK, I could add a time constant to the backlight so that it only comes on if the squelch remains open for a set time.

I've added a new option to enable/disable faster scanning - this also controls the faster squelch which causes the LED to be twitchy. See how things go with faster scanning disabled.

pawol, 2023-10-03:
The question is... how often you use scanning and how often you get angry because of self liting background and green led.

OneOfEleven, 2023-10-03:
My radio is always on channel scan.

pawol, 2023-10-03:
If it is an option then OK. By default that behaviour is bad an annoying :)

pawol, 2023-10-03:
I confirm that when fast scanning option is disabled then standard squelch works as expected. The ticket can be closed

closed 2023-10-03 by OneOfEleven