OneOfEleven/uv-k5-firmware-custom #91: Is it possible to remove FMRadio funciton but make some code change that user can input Radio freq directly in VFO mode to listen to broadcast, to reduce the size of firmware.

troilus, 2023-10-02:


OneOfEleven, 2023-10-02:
You like long titles ;)

OneOfEleven, 2023-10-02:
I suppose it could auto switch to FM Radio mode if frequency is within the broadcast band

troilus, 2023-10-02:
In my personal opinion, the search, save, etc. functions of the FM radio function are not so necessary, so I came up with this idea, wondering if it could reduce the firmware size.

troilus, 2023-10-02:
I'm sure it's late at your place, good night!

bricky149, 2023-10-02:
One thing I found out recently was that LTO builds w/o the FM radio option made binaries too small to flash. It took me by surprise as I usually compile with it included. Maybe I can look at it when I have time so 1o11 can focus on bug fixes.

OneOfEleven, 2023-10-02:
Too small ?  .. do they still work I wonder

lohtse, 2023-10-02:
> One thing I found out recently was that LTO builds w/o the FM radio option made binaries too small to flash. It took me by surprise as I usually compile with it included. Maybe I can look at it when I have time so 1o11 can focus on bug fixes.

what was the files size??? I used your brand  diasbled every but 
ENABLE_LTO                    := 1
ENABLE_UART                   := 1

this is what I got
  text    data     bss     dec     hex filename
  47664      12    2608   50284    c46c firmware

Oh and shitty AM quality lol

OneOfEleven, 2023-10-02:
I don't yet know why your AM sometimes works and the rest of the time don't. I may spot something that triggers my brain into seeing the problem, but so far I just don't know.

It's not as if the AM fix I've done is complicated, it isn't it, it does nothing more than adjust the front end RF gain according to the RSSI level - in order to stop the AM demodulator from saturating. That's it.

Maybe someone will spot something and let me know, who knows.

lohtse, 2023-10-02:
> I don't yet know why your AM sometimes works and the rest of the time don't. I may spot something that triggers my brain into seeing the problem, but so far I just don't know.
> 
> It's not as if the AM fix I've done is complicated, it isn't, it does nothing more than adjust the front end RF gain according to the RSSI level (maintains a constant RSSI level of -89dBm) - in order to stop the AM demodulator from saturating (the AM distortion everyone hears without the fix). That's it.
> 
> Maybe someone will spot something and let me know, who knows.

Latests builds have been good again...Apologies for not letting you know sooner..

ea7hjb, 2023-10-02:
The idea of the original FM Mode is thinking on the convenience of being able to listen to the music while the HT is monitoring the VFOs. If RX happens, the music is stop and you listen to your VFOs. After a while of silence, the system goes back to the music.

I think this is a great approach, and removing such behaviour would mean to have to manually tune the frequency, having the VFO permanently busy (forget about the fake Dual Watch). In my opinion, this is the best approach, as in a certain mode it manages to have a truly Dual VFO, even if it is only for the commercial FM.

OneOfEleven, 2023-10-03:
But ..

What if they want to use the 88-108 band but in normal FM comms mode ?

I thinking that as the VHF broadcast band is going to be closed down at some point to all broadcast FM (at least here in the UK), it's all going digital (DAB radio) only mode, the FM band will no longer support broadcast radio.

lohtse, 2023-10-03:
> But ..
>
> What if they want to use the 88-108 band but in normal FM comms mode ?
>
> I thinking that as the VHF broadcast band is going to be closed down at some point to all broadcast FM (at least here [...]

I thought the gov went back on that as DAB cannot be received everywhere  so would cut off folk as it would not allow them to get information out to them if they could not RX DAB...

OneOfEleven, 2023-10-03:
oh right :)

ea7hjb, 2023-10-03:
Here in my country, DAB is a joke, only working in a few places and not even broadcasting all station. Most likely DAB will never work at 100%, like in other countries, so removing FM mode is quite... a disconsideration for people living in countries in this situation. However I think there is a switch in the Makefile called "ENABLE_FMRADIO", so logic dictates that this should be the trigger that decides if the range 87-108 is to be input as another common frequency or used in the special FM radio mode.

pawol, 2023-10-04:
Yes, if needed then radio can be disabled during compilation.

bricky149, 2023-10-22:
May as well update this.

I integrated the FM radio stuff into the main frequency selection a while back but only saved a couple hundred bytes from the app code removal. Pulling in the BK1080 driver code is where most of the added size comes from.

I lost it when the forks diverged. 👉👈

closed 2023-10-24 by OneOfEleven