OneOfEleven/uv-k5-firmware-custom #92: PTT-ID

ea7hjb, 2023-10-02:
Hello, doing some test, found the PTT-ID thing is a bit glitchy. I don't know who uses this but, there it go:

If setting to:
OFF: well, it's OFF.
BOTH: Sends DTMF at beginning and ending of the transmisión.
KEY DOWN: Same as "BOTH"!
KEY UP: Doesn't work at all.

If you turn off and back on the Quansheng, the PTT-ID setting reverts back to OFF. That is, it must be enabled each time you power on. Is this by design?

I also noticed that sometimes if Roger Beer is activated, the beep/mdf may play simultaneosuly over the ending DTMF code, but I am not sure about this. The processing power of my brain at this hours of the night is low. Could someone check this? I think this does not always occur.

OneOfEleven, 2023-10-02:
It WAS like that a day or so ago but I fixed it, it should be fine now with the most recent firmware ?

The ROGER beeps and the DTMF shouldn't be over-lapping, the DTMF should come AFTER the ROGER/MDF - have a look on an audio recording in an audio editor or something. Let me know.

ea7hjb, 2023-10-02:
I assure you I was using the e0aa1a8 version, right to test the CTCSS bug you fixed (perhaps you can see the 210,7Hz in the video). Haven't tested 9ac3525 yet. I found things worked as expected until, suddenly, after changing the setting a hundred times, they didn't.

Regarding the overlapping roger beep with the DTMF codes, I can't bet all my money, but I could say I think it has happened. Will keep trying to provoke the failure

https://github.com/OneOfEleven/uv-k5-firmware-custom/assets/17039268/3c868c27-9df1-4a2e-a5bf-2a8db0564c75

.

OneOfEleven, 2023-10-02:
ah, you mean when it's set to TX-UP rather than BOTH. Just fixed that.

The ROGER and DTMF I've never seen overlap though, yours are perfect.

[image]

ea7hjb, 2023-10-02:
Well, Key Up sounds when releasing the button (I don't know if this is how it's mean to be, perhaps it would be better to use the original nomenclature, "BOT" (Begin of transmission) and "EOT" (End of transmissión? less confusing?).

Note the fails occurs when selecting Key DN. It transmit both at start/ending! Just like BOTH option!

Also note that in my first message, I said the Key UP didn't work at all, and I was not mistaken. Seems like the error is random. Perhaps the menu is not storing the proper value in the EEPROM? That would explain why sometimes you turn off and on the device and the option is set to OFF.

Regarding the roger beep, I would say that sometimes, if BOTH is set, and you quickly press/release the PTT, the roger beep will overlap either the start DTMF or the ending DTMF, depending on how fast your finger was. But I could not record this, so I will not insist until I can properly register the failure.

OneOfEleven, 2023-10-02:
The PTT-ID is configured per channel and per frequency band (7 bands).
The ROGER is a global setting which is independent of channel or band.

So if you move to different frequency band (1 of 7) the PTT-ID off/up/down/both will change, as it will with individual channels.

closed 2023-10-03 by OneOfEleven