OneOfEleven/uv-k5-firmware-custom #94: Wrong batt percentage on commit e0aa1a8

pawol, 2023-10-02:
Extremally wrong batt percentage on commit e0aa1a8
For instance I see 2 bars and 19% of battery. Battery voltage is 7.49V

I use battery callibration as below:
Level 0: [1296] 5.00 V // Empty battery blinking
Level 1: [1814] 7.00 V // 1 battery bar if above this value
Level 2: [1916] 7.40 V // 2 battery bars if above this value
Level 3: [1969] 7.60 V // 3 battery bars if above this value, also value used to calculate adc to volt
Level 4: [2046] 7.90 V // 4 battery bars if above this value
Level 5: [2300] 8.88 V // overwritten by radio to 2300 anyway

and ther the batt bars seems to be proper regarding batt voltage

So, when I have 7,49V on batt then we should rather see 2 bars and percentage near 40-50%.

7,49V is still quite good condition of the batt.

7.49/2=3,745 V per LiIon cell

OneOfEleven, 2023-10-02:
Ignore the battery bar level, that's always been very inaccurate. I'll put that back to original QS levels and let people rely mainly on the percentage for a better battery level indicator.

OneOfEleven, 2023-10-02:
I've restored the battery bar thresholds to that of the calibration values.

The battery percentage reading is  not based on those bars, the percentage is now as per [egzumer](https://github.com/egzumer)'s code, he did a detailed voltage measurement through the batteries full discharge cycle and from it created a more accurate percentage algorithm.

pawol, 2023-10-02:
Will check :)

pawol, 2023-10-02:
It seems that ENABLE_OVERLAY = 0 now and previously it was 1. What does it mean? Is it necessary to enable?

OneOfEleven, 2023-10-02:
> It seems that ENABLE_OVERLAY = 0 now and previously it was 1. What does it mean? Is it necessary to enable? I do not want to brick my QS.

If your that concerned about bricking it then please don't use this firmware, stay with Quanshengs own firmware.
All these open firmwares is experimental, we don't have proper access to proper datasheets to know exactly how to configure things the way we want.

The OVERLAY bit is to do with accessing the CPU's flash, it's actually safer to have that set to '0' (disable flash access from the firmware).

pawol, 2023-10-02:
Please do not exaggerate."I Do not want" means I want to avoid if it is possible because of missing OCD

ea7hjb, 2023-10-02:
For some reason, the firmware considers 8,14v is 100%, and anything above is also seen as 100%. As far as my knowledgement reach, the 100% should be 8.40v, which is the maximum for two ion cells.

If relevant, I must say calibrated the battery using OpenQuack and providing exactly 8,40v (during the screen on procedure, with a 4-point measurement and a lab power supply). The indications are now quite accurate. I could not use the egzumer method, because on the menu I could never select the voltage that was being feeded to the HT. It was always higher or lower. Doesn't matter if you alter the output of the PSU, it would always measure higher or lower, being unable to select the correct valure. I don't know was I was doing wrong.

OneOfEleven, 2023-10-02:
I haven't checked egzumer's settings, I just assumed they'd got it right. But yes 100% is 8.4V for the battery, or should be.

OneOfEleven, 2023-10-02:
Well, don't change the frequency cal without a good counter. It is in the hidden section to hide it.

The other settings however (TX bands) could get you taken away depending on where you live !

OneOfEleven, 2023-10-02:
But I'll add a compile option for the F-CAL no problem.

ea7hjb, 2023-10-02:
Great! Changing the range to pMin=1600 and pMax=2200 allowed for a very very fine adjust. Still need to fix the thing that 8,14v != 100%. But with time...

Thank you! And thank you for the "F-CAL". I think this is more safer, since that option is near to the "BATCAL" and one could, by mistake, touch it and mess it greatly.

OneOfEleven, 2023-10-02:
I put an adjustment range limit on the F-CAL when I wrote it, it's range limit depends on where the user is in frequency.

If you like playing radio, I'd say buy a nice little frequency counter, they are useful.

I'm putting a 2 minute menu time-out on the F-CAL and BATCAL menus, otherwise the menu timeout is 20 seconds (set near the top of misc.c).

ea7hjb, 2023-10-02:
I understand, and I have access (not my own) to very accurate Frequency Counters. That is why I said it would be sad to misalign by mistake the only thing the Quansheng comes properly aligned from factory. Because as you know, the voltage and the SQL are not very... fine. But the frequency it is. Quite a lot.

The 2 minutes timeout is a good idea, indeed. And, if this is about giving ideas... :P would it be possible (when you have time, of course, I know fixing bugs is more important) to add the posibility to do TX on the CTCSS/DCS menu? The old Yaesus had that feature, very usefull to quickly find the subtone for accessing a repeater, without having to re-enter the menu each time you pressed the TX. Many peope don't remember this and call me crazy, but it is true!

OneOfEleven, 2023-10-02:
> would it be possible to add the posibility to do TX on the CTCSS/DCS menu?
> The old Yaesus had that feature, very usefull to quickly find the subtone for accessing a repeater, without having to re-enter the menu each time you pressed the TX.

I've done that now. You can't use the up/down keys whilst TX'ing as they are for DTMF whilst TX'ing, but if you quickly de-key TX, press up/down to change CTCSS tone, then key TX back up it use the new CTCSS tone.

OneOfEleven, 2023-10-02:
Can also now very quickly find the scrambler inversion frequency of a transmission by changing the inversion freq in the menu whilst RX'ing the signal - couldn't do that before.

egzumer, 2023-10-06:
To fix wrong bar levels go to the BATCAL, enter and confirm the value without changing. The battery bar levels are recalculated on save. This was implemented recently.

As to 100% at 8.14V see https://github.com/egzumer/uv-k5-firmware-custom/issues/2

closed 2023-10-16 by OneOfEleven