OneOfEleven/uv-k5-firmware-custom #96: Turbo Scan 💯 *Not an issue*

M7OCM, 2023-10-02:
Wow the scan rate has risen dramatically thanks 111 :-)

PieeerB, 2023-10-02:
It remains to make a choice of the scanning range vfo1-vfo2

OneOfEleven, 2023-10-02:
> Wow the scan rate has risen dramatically thanks 111 :-)

So long as it's not to fast so as to miss signals, it takes time for the PLL to lock and for the squelch to open/respond, but I've been testing difference speeds to find the 'just to fast' threshold (around 50ms per channel is the fastest really)., then almost double the time spent (80ms actually) on each channel just to be sure it's OK.

OneOfEleven, 2023-10-02:
> It remains to make a choice of the scanning range vfo1-vfo2

Bug fixes before adding anything else I think.

M7OCM, 2023-10-02:
> So long as it's not to fast so as to miss signals, it takes time for the PLL to lock and for the squelch to open/respond, but I've been testing difference speeds to find the 'just to fast' threshold [...]

Yes understandable, hit a station straight away so seems to be perfect 111. Well done.

PieeerB, 2023-10-02:
are there any plans to make a scanning range??

OneOfEleven, 2023-10-02:
I'm sure someone might in time. I'm not sure how much longer I shall continue myself., I've done a lot to add too the project.

OneOfEleven, 2023-10-02:
I've only increased the channel scanning speed at the moment. I want to increase the frequency scan speed too (scanning across a frequency band).

lohtse, 2023-10-02:
> I've only increased the channel scanning speed at the moment. I want to increase the frequency scan speed too (scanning across a frequency band).

now that would be awesome!!!

DE8MSH, 2023-10-02:
Scanning by 1O11:

https://www.youtube.com/shorts/_YZS4t4G9Us

OneOfEleven, 2023-10-05:
Making it scan faster means making the squelch open/close faster in order to keep up, but the side effect is a twitchy squelch (flashy green LED).

closed 2023-10-05 by OneOfEleven