OneOfEleven/uv-k5-firmware-custom #98: Chirp Not work

Braccato, 2023-10-02:
Hi, with the latest updates there are dialogue problems with chirp.

OneOfEleven, 2023-10-02:
I'm afraid if people want extra/different menu options/settings then that's inevitable unfortunately :(

I've had a fair few 'I wants' from people, some I have implemented, some I haven't.

Do I not change the radio's config to keep it compatible with the PC softwares, or do I add/change stored features that makes the radio become incompatible with the PC softwares ?

It's a problem.

bricky149, 2023-10-02:
I think the original OEFW should retain compatibility with Chirp while this fork focuses on QoL improvements. I see no issue with Chirp incompatibility.

ik5wwp, 2023-10-02:
Problems with Chirps only during  upload from pc to radio. The radio switch in TX and the serial comunication collapses.
If you set a frequency where the TX is disabled (for example 40Mhz) the upload goes fine.

OneOfEleven, 2023-10-02:
oh right, interesting.

I'll have a look, I assumed all radio activity was disabled while a config upload/download was in progress. I'll make sure though.

lohtse, 2023-10-02:
> Hi, with the latest updates there are dialogue problems with chirp.

What Dialog issues? I use it daily (just did now) no issues..

lohtse, 2023-10-02:
> Problems with Chirps only during upload from pc to radio. The radio switch in TX and the serial comunication collapses. If you set a frequency where the TX is disabled (for example 40Mhz) the upload goes fine.

Has that not alway been the case???  So if so that would be an issue to raise with Chirp.

lohtse, 2023-10-02:
> I think the original OEFW should retain compatibility with Chirp while this fork focuses on QoL improvements. I see no issue with Chirp incompatibility.

Same here I pretty much use CJorp daily (just used it) and not had issues other than whats always plagued the K5 regardless of whether stock,UVmodded or OEFW...

c3kkos, 2023-10-03:
It is not firmware related, at least, not this one, not today.

Until eeprom data structure and offsets are preserved, the chirp driver will continue to work flawlessy.

Some users are reporting TX ON while serial data is connected but this is due to a flaw in k5 USB-C electrics, so defenitely a HW issue, not a SW one.

This said, there is A SUSPECT that the more the fw put work into mcu, the more the k5 become somehow sensible to RF induction/returns.

This can be seen as display can go crazy; many users have reported that with a vfo set at 145.100Mhz, doing TX while normally holding the k5 with the hand.

This is a well known SPI communication flaw; the more the data transmitted, the more the channel is busy, the more chance we have to encounter problems.

NOTE: all k5s are different. Personally i NEVER encountered such issues, but there are videos on telegram groups that clearly show that, so it's not a joke. Sadly, as explained elesewhere, there is no easy fix and probably never will.

As a very, very last note, i feel that the same "busy firmware problem" is somehow affecting the overall sensitivity of k5, but this is more a feeling and not a fact at all. Nonetheless, measurements has shown a 7/8dbm loss in RX between the firmwares based on OEFW, compared to the old ones made by Piotr022. Take a minute to test the piotr's v78 rssi bar firmware. That was the one seemingly most sensitive.

ik5wwp, 2023-10-03:
Maybe...
But with a FW Piotr 78 based on version 26 I have non problem of unwantend TX during upload from CHIRP.
How do you explain it?

OneOfEleven, 2023-10-03:
Yes the K5 does have a HW problem with RF getting back into where it's not wanted (display problem with hand on radio). As soon as I heard the details that was only conclusion to arrive at, any decent RF engineer will tell you tell you the same.

Anyway, I did put in a timer (on the serial comms) yesterday that prevents the TX being keyed up whilst there is any config upload/download going on. I presume that's not working as it should then ?

ik5wwp, 2023-10-03:
I know the RF problems of the K5, but the strange thing is that with a certain firmware Chirp work flawlessy, with this one it does not work. Now the situation is worse than before: "Bad response header" also when download from radio and not only in upload.
Maybe it's due to lenght of the timer mentioned above?

ik5wwp, 2023-10-03:
"Radio parameter reading failed" also with the origina QS Portable CPS radio software.

OneOfEleven, 2023-10-03:
No the timer I added has no effect on the comms, it purely disables the TX coming on due to the radio seeming the plugged in serial port looking like a PTT press.

If people look at the UART code they will see this.

OneOfEleven, 2023-10-03:
Yes I don't doubt there are now comms 'errors' seem by the PC software(s), people want/demand extra configuration options, they have to be stored in EEPROM, that is the radios config storage area, which is what is read/written by the PC software(s).

I can restore that back to original, but then you loose the extra options I've added/done, you'll have to re-configure those options in the menu manually every time you switch the radio on. It's an either/or thing I'm afraid.

I could as easily write new custom configuration PC software to replace chirp and the QS version, but I ain't going to, it's now a £9.70 (inc postage) radio on ali, it's a cheap toy for people to play with, my time is more precious to'**_me_** than that.

ik5wwp, 2023-10-03:
Please, keep in mind that I appreciate very much your work and the time you spend to do it.
Mine are only suggestions, not accuses!
Thank you again and anyway for what you do!

OneOfEleven, 2023-10-03:
No problem. It's the social media ke-nuckle draggers that have been escaping and finding me here that's causing me to go into defence mode ;)

ik5wwp, 2023-10-03:
I perfectly understand...

OneOfEleven, 2023-10-03:
Just having a look at the comms issue. The radio is rebooting shortly after a config read is started from the PC. Hoping it's not a stack overflow problem or such like.

OneOfEleven, 2023-10-03:
Try again with chirp, let me know what happens with latest firmware.

ik5wwp, 2023-10-03:
I will try again in the afternoon or maybe before.
Thank you!

ik5wwp, 2023-10-03:
Bingo! The problem seems to be solved.
Done many download/upload with no problem.
Thank you very much!

OneOfEleven, 2023-10-03:
Good good, thank you for letting me

OneOfEleven, 2023-10-03:
> As a very, very last note, i feel that the same "busy firmware problem" is somehow affecting the overall sensitivity of k5, but this is more a feeling and not a fact at all. Nonetheless, [...]

I've just done a test on 145MHz (can't test UHF as I've lost an attenuator I need) with this firmware and the original firmware, they are both the same on the signal generator.

In FM wide mode, they both detect a -118dBm carrier the same.

OneOfEleven, 2023-10-03:
Can you show me what's happening on a short video at all at a couple of different frequencies ?

closed 2023-10-03 by OneOfEleven