OneOfEleven/uv-k5-firmware-custom #102: F7 band locked to .600

Yag00000, 2023-10-03:
F7 band locked to .600

OneOfEleven, 2023-10-03:
Try it now, found a bug I made when changing to a simpler frequency band table yesterday.

pawol, 2023-10-03:
I confirm. Cannot put  eg. 470 MHz into VFO

OneOfEleven, 2023-10-03:
I can

OneOfEleven, 2023-10-03:
This isn't egzumers firmware repo though

pawol, 2023-10-03:
OK. He is behind you a few commits

lohtse, 2023-10-03:
> This isn't egzumers firmware repo though

I think the reason he keeps raising issues here is Egzumer has stated on telegraph he just merges  yours and Fagci's FW together and nothing more is thats the case anything unrelated to the spectrum is related to your FW.. And I can confirm the issue.. BUT egzumer  will not be merging the commits for  at least a few hours  yet.   having said that would still be best to start the ssue with his repo first.

pawol, 2023-10-03:
On the latest 1of commit bug is fixed - I confirm

closed 2023-10-03 by OneOfEleven