OneOfEleven/uv-k5-firmware-custom #108: icon COMPANDER TX RX still not show in screen when is active

Yag00000, 2023-10-04:


Yag00000, 2023-10-04:
icon COMPANDER TX RX still not show in screen when is active

lohtse, 2023-10-04:
you do know Compander is not working properly yet and is as buggy as hell?

Yag00000, 2023-10-04:
in 1of11 works fine in TX mode only, not RX

lohtse, 2023-10-04:
> in 1of11 works fine in TX mode only, not RX

Theres quite a lot of discussion about it on telegram and many  devs suggest not even enabling it   even 1of11 says this

ENABLE_COMPANDER              := 0       this is broken, don't bother with it for now .. compander option (per channel)

OneOfEleven, 2023-10-04:
I'm currently trying the problem in RX mode, no audio from it for some reason.

But I think the OP is talking about just the on-screen indicator (little 'C' character) that should show when the compander is enabled. I'll have a look and see why that might be.

[image]

OneOfEleven, 2023-10-04:
I can't imagine why the little symbol shows for some but not others.

wrcrooks, 2023-10-24:
@Yag00000 Is this still an issue on the latest firmware?

closed 2023-10-28 by OneOfEleven