OneOfEleven/uv-k5-firmware-custom #109: SUGGESTION ONLY... scanlists

lohtse, 2023-10-04:
Is it possible to ad a short cut key to toggle between scanlist  1 & 2???  something similar to the A/B button ut may be using  and f key combo..

OneOfEleven, 2023-10-04:
It's been there for quite some time - long press key-5 on selected channel, it toggles scanlist 1 & 2.

I'll add the option to compile out VOX first (as per the other post).

lohtse, 2023-10-04:
> It's been there for quite some time - long press key-5 on selected channel, it toggles scanlist 1 & 2.
> 
> I'll add the option to compile out VOX first (as per the other post).

hmmmm nothing happens for me when I do that though..

OneOfEleven, 2023-10-04:
oh.

Are you doing it on a channel ? .. rather than a VFO/frequency

OneOfEleven, 2023-10-04:
Have to got NOAA enabled in Makefile ?

lohtse, 2023-10-04:
> Have to got NOAA enabled in Makefile ?

Nope always disabled

OneOfEleven, 2023-10-04:
I'll have a look later then, off out shortly.

It's always been there though, I use it all the time. I'll see what might b e wrong later.

lohtse, 2023-10-04:
> I'll have a look later then, off out shortly.
> 
> It's always been there though, I use it all the time. I'll see what might b e wrong later.

Please don't rush etc just get out an enjoy the day.... I didn't even know it was available so I can't miss what I didn't know about...

OneOfEleven, 2023-10-04:
No I didn't advertise it very well, my bad.

lohtse, 2023-10-04:
> No I didn't advertise it very well, my bad.

Did you mean you add the ability to add to scanlists??? as that works by long press 5

But I mean actually toggling the actually scan list in use.. so if you want to scan list one it scans list one  but if you want to scan list two you can toggle two instead of going through the menus.

CaptainRi, 2023-10-04:
> 它已经存在相当长一段时间了 - 在选定的频道上长按 key-5，它会切换扫描列表 1 和 2。
> 
> 我将添加首先编译 VOX 的选项（按照另一篇文章）。

> 它已经存在相当长一段时间了 - 在选定的频道上长按 key-5，它会切换扫描列表 1 和 2。
> 
> 我将添加首先编译 VOX 的选项（按照另一篇文章）。

Do you mean to plan to adjust the shortcut key for s-add from f5 to f7

OneOfEleven, 2023-10-04:
> Did you mean you add the ability to add to scanlists??? as that works by long press 5
>
> But I mean actually toggling the actually scan list in use.. so if you want to scan list one it scans list one [...]

You want to switch between the scan lists with a key press ? .. with the VOX now disableable yes I can use long/func key-7 for that.

I need to add the new long-press/function key list to the readme, otherwise no one will know about them.

OneOfEleven, 2023-10-04:
I can see the long press menu key being a problem ;)  .. I will extend the 'long' time for that key in particular.

lohtse, 2023-10-04:
> You want to switch between the scan lists with a key press ? .. with the VOX now disableable yes I can use long/func key-7 for that.
>
> I need to add the new long-press/function key list to the [...]

Thank you.....

take your time its not important just usefull. so there is no hurry what so ever.

OneOfEleven, 2023-10-04:
I've done the scanlist toggle.

You start scanning as per normal with long-press-*-scan key, then to do the same to toggle between scanlists.

To cancel scanning press the exit key.

lohtse, 2023-10-04:
> I've done the scanlist toggle.
> 
> You start scanning as per normal with long-press-*-scan key, then to do the same to toggle between scanlists.
> 
> To cancel scanning press the exit key.

AWESOME works a damn treat... thank you

OneOfEleven, 2023-10-04:
I've been trying to get it to include the other VFO into the scanning if dual-watch is enabled. Bare with me on that one.

OneOfEleven, 2023-10-04:
You can now set it to ALL channels as well as scanlist-1 or 2

closed 2023-10-05 by OneOfEleven