OneOfEleven/uv-k5-firmware-custom #115: Copy channel to vfo does not work in 429ac36

pawol, 2023-10-04:
Copy channel to vfo does not work in 429ac36.
I have freq+name on dusplay A and B. Long pressing of M does not copy frewuency into vfo.

INDIAJULIETVICTOR, 2023-10-04:
Confirm..

https://github.com/OneOfEleven/uv-k5-firmware-custom/assets/145300322/775c81e2-b7de-4a1d-8b5c-fd01ed6075ff

OneOfEleven, 2023-10-04:
I've just changed how it works. It now takes the current channel, sets the VFO to frequency mode and copies the channel into it (doesn't effect the other VFO).

Have a play

closed 2023-10-04 by OneOfEleven

INDIAJULIETVICTOR, 2023-10-04:
not again..

https://github.com/OneOfEleven/uv-k5-firmware-custom/assets/145300322/61890672-b4bf-41de-a093-f84b7d8eee14

INDIAJULIETVICTOR, 2023-10-04:
I found...the DUAL WATCH OPTION

`
				if (gEeprom.VFO_OPEN &&
				    // gEeprom.DUAL_WATCH == DUAL_WATCH_OFF &&
				    gScanState == SCAN_OFF &&
				    gCssScanMode == CSS_SCAN_MODE_OFF)
				{	// copy channel to VFO

`