OneOfEleven/uv-k5-firmware-custom #117: tone 1750Hz does not open squelch and stops list scanning on 429ac36

pawol, 2023-10-04:
tone 1750Hz does not open squelch and stops list scanning on 429ac36 .
When the tone 1750 is send by 1st QS then squelch is not opened on 2nd WS 
Same issue when 2nd QS scans list containing proper channel.

OneOfEleven, 2023-10-04:
I thought the 1750Hz tone is for opening old style ham repeaters, not other radios. 1750Hz is the repeater tone access frequency.

OneOfEleven, 2023-10-04:
But the 1750Hz method is all but gone now, it's all CTCSS.

pawol, 2023-10-04:
air is not important what opens squelch. We have to think about tone 1750 as common audio. Does not mater 1000Hz or 2000Hz. Modulater RF is send and has to open each standard squelch.

OneOfEleven, 2023-10-04:
These radios are aimed at the radio ham market, which is the reason for the 1750Hz repeater access tone frequency.

pawol, 2023-10-04:
No, 1750Hz is specified for some repeaters with squelch opened by frequency.
Like I said, in our case it does not matter what we transmit to open classical squelch. Fact is that the squelch is not opened nut should.

pawol, 2023-10-04:
I found now that squelch is not opened on pmr1 channel but on pmr2 works ikay. strange. Need more time to check a few thing. Maybe somerhing wring I have in pmr1 channel config

pawol, 2023-10-04:
Completly strange issue. After a while pmr2 stoped working also.

pawol, 2023-10-04:
Maybe this bug affect sending dtmf string with missing 1st character also - PR was reported earlier. 1st hight amplitude dtmf cannot open squelch and after pause the 2nd tone works okay. In this case I send one high amplitude and continuous signal. That is my theory.

pawol, 2023-10-04:
That is it! Fast scanning impact on squelch in this case and in DTMF case (PR #72)
On normal scanning option everything works okay - squelch opened by 1750 and missing 1st character in DTMF string

pawol, 2023-10-05:
The issue fixed on e042b21. PR closed

closed 2023-10-05 by pawol