OneOfEleven/uv-k5-firmware-custom #120: A slight issue

lohtse, 2023-10-04:
just tried compiling new build I got thisn 

irmware-custom//external/CMSIS_5/Device/ARM/ARMCM0/Include -c ui/main.c -o ui/main.o
ui/main.c: In function 'UI_DisplayAudioBar':
ui/main.c:94:4: error: this 'if' clause does not guard... [-Werror=misleading-indentation]
   94 |    if (gCurrentFunction != FUNCTION_TRANSMIT ||
      |    ^~
ui/main.c:104:5: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the 'if'
  104 |     const unsigned int voice_amp  = BK4819_GetVoiceAmplitudeOut();  // 15:0
      |     ^~~~~
cc1.exe: all warnings being treated as errors
make: *** [ui/main.o] Error 1
Press any key to continue . . .

OneOfEleven, 2023-10-05:
Is it still happening I wonder ?

lohtse, 2023-10-05:
> Is it still happening I wonder ?

yours compiles fine now... other forks fail if tx1750 disabled

closed 2023-10-05 by lohtse