OneOfEleven/uv-k5-firmware-custom #124: Potential value underflows in keyboard.c

bricky149, 2023-10-05:
I've been running a few code warnings to help with spotting issues and I found this while running with -Wsign-conversion:

```
driver/keyboard.c:46:29: warning: unsigned conversion from 'int' to 'short unsigned int' changes value from '-1' to '65535' [-Wsign-conversion]
   46 |         .set_to_zero_mask = ~(0),
      |                             ^
driver/keyboard.c:57:29: warning: unsigned conversion from 'int' to 'short unsigned int' changes value from '-1025' to '64511' [-Wsign-conversion]
   57 |         .set_to_zero_mask = ~(1 << GPIOA_PIN_KEYBOARD_4),
      |                             ^
driver/keyboard.c:67:29: warning: unsigned conversion from 'int' to 'short unsigned int' changes value from '-2049' to '63487' [-Wsign-conversion]
   67 |         .set_to_zero_mask = ~(1 << GPIOA_PIN_KEYBOARD_5),
      |                             ^
driver/keyboard.c:77:29: warning: unsigned conversion from 'int' to 'short unsigned int' changes value from '-4097' to '61439' [-Wsign-conversion]
   77 |         .set_to_zero_mask = ~(1 << GPIOA_PIN_KEYBOARD_6),
      |                             ^
driver/keyboard.c:87:29: warning: unsigned conversion from 'int' to 'short unsigned int' changes value from '-8193' to '57343' [-Wsign-conversion]
   87 |         .set_to_zero_mask = ~(1 << GPIOA_PIN_KEYBOARD_7),
      |                             ^
```

Is this intentional? I understand @wutje rewrote this recently to save space.

wutje, 2023-10-05:
No that was not intended. But it probably does not matter. It is used as a binary mask so the sign bit is not relevant. If needed I can provide a fix. But not before the weekend.

bricky149, 2023-10-05:
There's a fix up already. Well done, 1o11!

OneOfEleven, 2023-10-05:
Just uploaded a fix that may clear those particular warnings.

It's nice not to see warnings, I'm all for one for making the compile output as easy to read as possible.

OneOfEleven, 2023-10-05:
I just happen to be at the laptop as you posted ;)

lohtse, 2023-10-05:
> I just happen to be at the laptop as you posted ;)

I think they just broke it lol
[image]

lohtse, 2023-10-05:
arm-none-eabi-gcc  -Os -Wall -Werror -mcpu=cortex-m0 -fno-builtin -fshort-enums -fno-delete-null-pointer-checks -std=c11 -MMD -flto=2 -DPRINTF_INCLUDE_CONFIG_H -DGIT_HASH=\"6b44659\" -DENABLE_FMRADIO -DENABLE_UART -DENABLE_SMALL_BOLD -DENABLE_KEEP_MEM_NAME -DENABLE_WIDE_RX -DENABLE_TX_WHEN_AM -DENABLE_F_CAL_MENU -DENABLE_CTCSS_TAIL_PHASE_SHIFT -DENABLE_SHOW_CHARGE_LEVEL -DENABLE_REVERSE_BAT_SYMBOL -DENABLE_CODE_SCAN_TIMEOUT -DENABLE_AM_FIX -DENABLE_AM_FIX_SHOW_DATA -DENABLE_SQUELCH_MORE_SENSITIVE -DENABLE_RSSI_BAR -DENABLE_AUDIO_BAR -DENABLE_COPY_CHAN_TO_VFO  -I D:/TWatch/uv-k5-firmware-custom/ -I D:/TWatch/uv-k5-firmware-custom//external/CMSIS_5/CMSIS/Core/Include/ -I D:/TWatch/uv-k5-firmware-custom//external/CMSIS_5/Device/ARM/ARMCM0/Include -c driver/i2c.c -o driver/i2c.o
arm-none-eabi-gcc  -Os -Wall -Werror -mcpu=cortex-m0 -fno-builtin -fshort-enums -fno-delete-null-pointer-checks -std=c11 -MMD -flto=2 -DPRINTF_INCLUDE_CONFIG_H -DGIT_HASH=\"6b44659\" -DENABLE_FMRADIO -DENABLE_UART -DENABLE_SMALL_BOLD -DENABLE_KEEP_MEM_NAME -DENABLE_WIDE_RX -DENABLE_TX_WHEN_AM -DENABLE_F_CAL_MENU -DENABLE_CTCSS_TAIL_PHASE_SHIFT -DENABLE_SHOW_CHARGE_LEVEL -DENABLE_REVERSE_BAT_SYMBOL -DENABLE_CODE_SCAN_TIMEOUT -DENABLE_AM_FIX -DENABLE_AM_FIX_SHOW_DATA -DENABLE_SQUELCH_MORE_SENSITIVE -DENABLE_RSSI_BAR -DENABLE_AUDIO_BAR -DENABLE_COPY_CHAN_TO_VFO  -I D:/TWatch/uv-k5-firmware-custom/ -I D:/TWatch/uv-k5-firmware-custom//external/CMSIS_5/CMSIS/Core/Include/ -I D:/TWatch/uv-k5-firmware-custom//external/CMSIS_5/Device/ARM/ARMCM0/Include -c driver/keyboard.c -o driver/keyboard.o
driver/keyboard.c:57:29: error: conversion from 'unsigned int' to 'short unsigned int' changes value from '4294966271' to '64511' [-Werror=overflow]
   57 |         .set_to_zero_mask = ~(1u << GPIOA_PIN_KEYBOARD_4),
      |                             ^
driver/keyboard.c:67:29: error: conversion from 'unsigned int' to 'short unsigned int' changes value from '4294965247' to '63487' [-Werror=overflow]
   67 |         .set_to_zero_mask = ~(1u << GPIOA_PIN_KEYBOARD_5),
      |                             ^
driver/keyboard.c:77:29: error: conversion from 'unsigned int' to 'short unsigned int' changes value from '4294963199' to '61439' [-Werror=overflow]
   77 |         .set_to_zero_mask = ~(1u << GPIOA_PIN_KEYBOARD_6),
      |                             ^
driver/keyboard.c:87:29: error: conversion from 'unsigned int' to 'short unsigned int' changes value from '4294959103' to '57343' [-Werror=overflow]
   87 |         .set_to_zero_mask = ~(1u << GPIOA_PIN_KEYBOARD_7),
      |                             ^
cc1.exe: all warnings being treated as errors
make: *** [driver/keyboard.o] Error 1
Press any key to continue . . .

OneOfEleven, 2023-10-05:
lol, what a fussy compiler you have !

Hold on then

bricky149, 2023-10-05:
> lol, what a fussy compiler you have !
> 
> Hold on then

It's not the compiler, I was going through a few -W flags to see what the compiler was silently changing. It's me being pedantic. ^^;

lohtse, 2023-10-05:
> lol, what a fussy compiler you have !
> 
> Hold on then

sooooooooooooooooooooooooooooorry lol

bricky149, 2023-10-05:
Thanks again for looking at this, 1o11.

I've got one more but it's for a feature that we know doesn't work yet:

```
settings.c:265:31: warning: unsigned conversion from 'int' to 'uint8_t' {aka 'unsigned char'} changes the value of '-49' [-Wsign-conversion]
  265 |                 Attributes &= ~MR_CH_COMPAND;  // default to '0' = compander disabled
      |                               ^
```

I'll close this as it's taking away from more important issues.

closed 2023-10-05 by bricky149

OneOfEleven, 2023-10-05:
it does work !

OneOfEleven, 2023-10-05:
OK, try that, give it all you got (-Wextra) !

OneOfEleven, 2023-10-05:
tut, I've broke the menu key now :(

bricky149, 2023-10-05:
> OK, try that, give it all you got (-Wextra) !

I think the work you've been doing in improving the functionality of this toy radio is more important than a few compiler warnings that only matter to software nerds like me. Maybe later when the code is more stable, for now I don't want to add extra work that can be avoided. :)

OneOfEleven, 2023-10-05:
All fixed :)

Just one line, a single number I got wrong.