OneOfEleven/uv-k5-firmware-custom #126: Proposals

INDIAJULIETVICTOR, 2023-10-05:
[PROPOSALS]

1 - instead of using the spectrum, which personally doesn't convince me much for several reasons, I think a function could be useful is a fast scanning between a range of frequencies, Freq Min and Freq Max (for example between the VFO up and VFO down).

2 - to avoid the password at boot by select on the makefile ( someone has a radio lock problem after flashing firmwares )

3  - in dual watch active when the Scan on memories is active to listen also the VFO, so you can watch both CHANNELS.

I know I'm a problem for you, please don't kill me.. I'm your most faithful follower.  :-D

OneOfEleven, 2023-10-05:
I've been testing including the other VFO for a bit now when channel scanning, not yet happy with it though.

I've added the power-on password option in Makefile.

The frequency spectrum/bandscope is something I'm definitely adding though (it's the reason I got into modifying the code to start with), many radio hams/scanner nerds love such tools as it lets you see at a glance what's happening on and around a desired frequency.

OneOfEleven, 2023-10-05:
If you do disable the power-on password code, it will wipe any password you may have set when you next turn the radio off - useful for anyone who forgets their password.

INDIAJULIETVICTOR, 2023-10-05:
Yes, I'm seeing that spectrum is in great demand... unfortunately I'm against it as there is greater use of the memory and processor and I wouldn't want all this use of the communication lines between the beken and the display to lead to a RF noise that disturbs the reception of weak signals.
It's my hypothesis, let's be clear.

OneOfEleven, 2023-10-05:
The bandscope code only runs when the user wants it, it won't have any effect at all on any of the various data lines on the PCB when the user doesn't have it running.

INDIAJULIETVICTOR, 2023-10-05:
exactly, I'm referring to when it comes into operation... how much memory space does it take to insert the bandscope?

OneOfEleven, 2023-10-05:
A fair bit, but it will have a compile option, you decide whether to include it or not.

INDIAJULIETVICTOR, 2023-10-05:
yes, this is the better solution..I would regret that some important functions could not be implemented due to this component.

closed 2023-10-28 by OneOfEleven