OneOfEleven/uv-k5-firmware-custom #128: DTMF issue

Romashca, 2023-10-06:
https://github.com/OneOfEleven/uv-k5-firmware-custom/assets/83151488/1429185a-f94f-4e66-bc4b-e7e655f57d3b

Romashca, 2023-10-06:
https://github.com/OneOfEleven/uv-k5-firmware-custom/assets/83151488/588e7e24-65be-41b3-8575-0dfd690e583b

Romashca, 2023-10-06:
the problem with dtmf, when pressing ptt 1 and # transmits the same tone as befwhen typing dmf and pressing ptt, nothing happensore pressing #

lohtse, 2023-10-06:
> the problem with dtmf, when pressing ptt 1 and # transmits the same tone as befwhen , when typing dmf and pressing ptt, nothing happens

could this be related/linked too your other issue???   I don't use dtmf.  and don't have the other issue. hmmmmmmmmmmmmmm

Romashca, 2023-10-06:
perhaps they are somehow connected? I need a dm to control the repeater

Romashca, 2023-10-06:
there are no such problems in the repo from 20231005

OneOfEleven, 2023-10-06:
Yes I broke the keypad stuff yesterday when I made a change, been on the case since tracking a couple of uncovered problems with the keypad code.

Romashca, 2023-10-06:
I really hope that you will succeed. We will be very grateful to you

OneOfEleven, 2023-10-07:
Still have a bug in the DTMF entry before TX.

If I enter some DTMF keys in, then stop and let the entry time out (go off screen), the next button press is ignored - that's a bug I MUST find.

Romashca, 2023-10-07:
you are my hero

closed 2023-10-10 by OneOfEleven