OneOfEleven/uv-k5-firmware-custom #129: TX issue

Romashca, 2023-10-06:
https://github.com/OneOfEleven/uv-k5-firmware-custom/assets/83151488/34757f49-7723-46ed-97ea-57107a6ea97d

lohtse, 2023-10-06:
could you actually explain the issue incluidng what bands/frqs please

The does happen  if you have just been using an AIRBAND frq  as takes a second or two to demodulate and switch back... this effects ALL fw.s including stock

i

Romashca, 2023-10-06:
There is actually no time difference after switching ranges. There is no such problem in the previous firmware version. this happens after the transition from A to B , or from the vfo

lohtse, 2023-10-06:
> There is actually no time difference after switching ranges. There is no such problem in the previous firmware version. this happens after the transition from A to B , or from the vfo

yeah you still did not provide the info I asked about so whatever

Romashca, 2023-10-06:
There is actually no time difference after switching ranges. There is no such problem in the previous firmware version. this happens after the transition from A to B , or from the vfo . Transition from VHF or UHF band

lohtse, 2023-10-06:
> There is actually no time difference after switching ranges. There is no such problem in the previous firmware version. this happens after the transition from A to B , or from the vfo . Transition from VHF or UHF band

Well I certainly dono't have the same issue. when On ops, we hav two channels  in use one is team which is in 400 range other is op controls which is in the 120-158 range and NEVER engounterd the issue.. and whats more I was doing this two hours ago

lohtse, 2023-10-06:
And for shits n giggles put in a similar frq to what you mentioned above and no issues.

Romashca, 2023-10-06:
I didn't mean to offend you in any way, I'm just trying to explain the problem, when switching from wuas, the same problem

lohtse, 2023-10-06:
https://github.com/OneOfEleven/uv-k5-firmware-custom/assets/48331155/357b6ea3-fd0e-4afb-b8d1-d525411b9bcf

vidoe may not show is well as I hoped bjut as I stated no issues here for me as you describe

Romashca, 2023-10-06:
perhaps there is some problem with the compilation of the firmware

lohtse, 2023-10-06:
> perhaps there is some problem with the compilation of the firmware

Possibly. try again... But I am NOT saying it is not a bug.. but could also be related to the compilation setup. Some settings may do odd things... I literally wanted to rule out an actual TX bug on those bands hence why I asked for that info... If they can be ruled out then you look elsewhere.

Romashca, 2023-10-06:
I repeated the compilation, there is no change. I turn on both channels at 145 mhz with a difference of 25 khz, after five minutes of downtime, it transmits on another channel instead of the dedicated one

Romashca, 2023-10-06:
[image]
the settings I used

lohtse, 2023-10-06:
are setups a re somewhat different you do have a few things enable I don't as I don't need them

 I think you will need to  just wait until 1of11 is able to review the code for a possible isse.

Romashca, 2023-10-06:
there are no such problems in the repo from 20231005  You are right, I am very grateful to you

lohtse, 2023-10-06:
please that it got solved for you

OneOfEleven, 2023-10-06:
I broke it

Bare with me

Braccato, 2023-10-06:
The same problem exists on my radio and a bug needs to be fixed

lohtse, 2023-10-06:
> The same problem exists on my radio and a bug needs to be fixed

DON'T DEMAND. just politely ask!!!!!!!!!!!!!!!!!!

OneOfEleven, 2023-10-06:
Not doing yet, I'm chilling today.

You must bare in mind that this is NOT a service being provided here, I'm NOT here to serve anyone but myself, I'm sharing purely out of the goodness of my heart, I work on it when and where the feeling takes me.

OneOfEleven, 2023-10-06:
> The same problem exists on my radio and a bug needs to be fixed

So fix it. I'm assigning this one to YOU. It's your turn this time.

Braccato, 2023-10-06:
we are here to report bugs, if we don't test the radios how do you fix the bugs? if we write here you don't have to get angry

Braccato, 2023-10-06:
you're wrong, I'm Italian and I use a translator, I just reported the bug, you're wrong.

Romashca, 2023-10-06:
Очень надеюсь, что у вас все получится. Мы будем вам благодарны

Braccato, 2023-10-06:
> Anche sul telegramma quando hai usato la lingua madre hai ancora parlato con un diritto che devi fare questo ora atteggiamento amico...
> 
> Ora in futuro si prenda cura...
> 
> I Devs fanno quello che fanno di per persona stessi PRIMA e risolvino le cose come e quando non sono possono che sono che fanno lo nel LORO tempo libero e senza costo per QUALSIASI .. Vieni a fare la storia, decidere cosa è importante risolvere e QUANDO....
> 
> Se è importante così tu. anche.Poi a imparare a codificare e da solo.. - Sono in modo semplice..

you are the one who made me delete from telegram you are hostile towards users you are a loser, you bring hate.

lohtse, 2023-10-06:
> you are the one who made me delete from telegram you are hostile towards users you are a loser, you bring hate.

And yet no one else complains other than you.. And infact I am constantly there heping others and being thanked...

You just didn't like being questioned when you started demanding things be fixed..... the same as you have done HERE...

lohtse, 2023-10-06:
> Not doing yet, I'm chilling today.
> 
> You must bare in mind that this is NOT a service being provided here, I'm NOT here to serve anyone but myself, I'm sharing purely out of the goodness of my heart, I work on it when and where the feeling takes me.

Suggest you close this thread  before it gets worse... and may be make other considerations etc ;-)

closed 2023-10-06 by OneOfEleven