OneOfEleven/uv-k5-firmware-custom #131: [Feature Request] Custom channel name by built-in keyboard.

MaiLunJiye, 2023-10-06:
In the original firmware, uvk5 just only can save channel with the name 'CH-XXX'.
If we want to custom channel name, we have to write channel by PC, which is so inconvenient.

OneOfEleven, 2023-10-06:
Yes that's right, I tried the original QS firmware back in the radio the other day, was somewhat surprised that I had great difficulty using it, just couldn't get the hang of it.

lohtse, 2023-10-06:
- [ ] I did that to show a colleague  the differnce and my head screamed at me.. within seconds was back on custom..felt so much better.

The name on device is a good idea and one that would be useful at somepoint in the distant future...

OneOfEleven, 2023-10-06:
yes lol.

Anyway, it's there, it's been there from almost the beginning.

lohtse, 2023-10-06:
> yes lol.
> 
> Anyway, it's been there from almost the beginning .. 

Ok now am confused..

How does it even work??

OneOfEleven, 2023-10-06:
1. Press Menu button
2. Scroll to "CH NAM" (around number 17)
3. Press the Menu button to enter
4. Use up/down keys to choose the desired channel
5. Press the Menu button again to enter edit name mode
6. Use the up/down keys to cycle through the letters etc
7. Press the Menu button again to move to the next character position
8. Repeat steps 5 and/or 6 till you reach the end
9. Press Menu button to save

Press the Exit button at any time to cancel the edit and return to the main menu.

Sounds a lot but once you done it a few times it will seem easy - hopefully.

OneOfEleven, 2023-10-06:
I'll put that in the readme.md file.

OneOfEleven, 2023-10-06:
Yes it only edits the name, so it only works on channels that are already there.

MaiLunJiye, 2023-10-07:
I'm sorry for my carelessness. Thank you for implementing this life change function

closed 2023-10-07 by MaiLunJiye

OneOfEleven, 2023-10-07:
No it's fine, don't worry.