OneOfEleven/uv-k5-firmware-custom #133: The radio transmits randomly on bottop or top channel.

egzumer, 2023-10-06:
It doesn't matter which is the main channel. It looks like it takes the one that the dual watch is currently on.

OneOfEleven, 2023-10-06:
I think I've fixed it now. Let me know.

I've also noticed a problem with the high/mid/low TX power as well, there appears to be only 1dB difference between them.

I don't have have a power meter here at the moment so can't check for, just a spectrum analyser without a suitable inliine attenuator.

egzumer, 2023-10-06:
Nope, still transmits randomly

egzumer, 2023-10-06:
From the tests that I saw on youtube the power levels did not correspond to the levels set with L/M/H, on some bands more than the others, but for some the power output did not change between for example L and M.

OneOfEleven, 2023-10-06:
Try now

OneOfEleven, 2023-10-06:
The TX power level is adjustable, but the stored calibration values seem way off :(

OneOfEleven, 2023-10-07:
A good H/M/L power setting would be something like 4W, 1W and 0.25W, which would be a much better dB ratio change, or even 4W, 0.5W and 100mW.

Currently 1dB difference is worthless/pointless.

Setting the actual output power is difficult though in practice, because as you say it's frequency dependent (very) and radio dependent, each individual radio needs it's own calibration table (which it already does have), but the user would need an RF power meter to calibrate there own radio(s).

joecotton, 2023-10-07:
> If the radios had an ADC reading sampling the level of RF TX output, I could add auto-calibrate code into the firmware, but I don't think the radios do have that feedback measurement do they ?

No, not typically. They would have to have a transformer to tap a little bit of the output RF, rectify it to DC, and feed into an ADC. No need to add that hardware for what would be a one-time use in the factory.

zhaoparrot, 2023-10-07:
> A good H/M/L power setting would be something like 4W, 1W and 0.25W, which would be a much better dB ratio change, or even 4W, 0.5W and 100mW.
> 
> Currently 1dB difference is worthless/pointless.
> 
> Setting the actual output power is difficult though in practice, because as you say it's frequency dependent (very) and radio dependent, each individual radio needs it's own calibration table (which it already does have), but the user would need an RF power meter to calibrate there own radio(s).
> 
> If the radios had an ADC reading sampling the level of RF TX output, I could add auto-calibrate code into the firmware, but I don't think the radios do have that feedback measurement do they ?

I also think the 4 1 0.25 arrangement is more reasonable. does this radio have battery current feedback? i may have seen this in some code. if so, u may use current to do a auto calibrate.

OneOfEleven, 2023-10-07:
It's possible to use current I spose. All I get is an ADC value, no idea on what the scale value is to convert that to say mA, though that doesn't matter.

I'll do a measurement and see what the current reading is for different output power levels (into a 50R load).

OneOfEleven, 2023-10-08:
Unfortunately, the current sensor doesn't return the current being used by the radio, it only seems to return a reading of some sorts when the radio is charging from the **USB** port, doesn't register desk/slot-in charger current.

zhaoparrot, 2023-10-09:
> Unfortunately, the current sensor doesn't return the current being used by the radio, it only seems to return a reading of some sorts when the radio is charging from the **USB** port, doesn't register desk/slot-in charger current.

I also did some test and confrim the **gBatteryCurrent** is useless, :(   random wiggle from 20 to 23 during TX

OneOfEleven, 2023-10-09:
I'm wondering if the ADC reading might actually be a voltage reading on the USB charge socket rather than a current reading, or something like that.

zhaoparrot, 2023-10-10:
I think the adc reading was using PA14 ，tracking the circuit board to this two holds near the USB-C port ,can't figure out which one .
[image]
I have try charge it with a usb power meter. 
when battery is 50% I got 5V 1A reading, gBatteryCurrent is 892±3
battery 94% 5V0.55A gBatteryCurrent is 861±3
still charging now.
 I was wondering what the gBatteryCurrent  would be if I unplugged the lithium battery while the USB cable was connected, but the user manual explicitly says not to do this. . .

zhaoparrot, 2023-10-10:
when the battery is fully charged  gBatteryCurrent still have 860 readings, but jumping  ,more stable when I disable batsav . maybe should rename this **gBatteryCurrent** to **gUSBCurrent**

OneOfEleven, 2023-10-10:
> I was wondering what the gBatteryCurrent would be if I unplugged the lithium battery while the USB cable was connected, but the user manual explicitly says not to do this. . .

Have to be careful about not having batteries installed, some designers wrongly rely on them as part of the charge/power circuit (basically as large capacitors) that smooth out voltage and current ups and downs.

OneOfEleven, 2023-10-10:
But knowing the what the circuit really is around all that would totally explain it all.

closed 2023-10-24 by OneOfEleven