OneOfEleven/uv-k5-firmware-custom #134: What's the point of modification of the "T VFO" function?

egzumer, 2023-10-06:
I can't find the issue that started that if any.

It doesn't make sense to me now, there is:
- MAIN - always transmit on the one that is chosen (filled arrow)
- UPPER/LOWER - always transmit on the UPPER/LOWER, which is exactly the same as MAIN, when you use A/B button the UPPER changes to LOWER and vice versa.

The main should be rather called RESPOND, and should transmit on the channel that the last transmission was received, so the user could respond to the caller.

ea7hjb, 2023-10-06:
The (Fake) Dual Watch of this device is quite... confusing. The Dual VFO, if active (doesn't matter if main is Upper or Lower VFO) will quickly switch between the two searching an active conversation, and in case of RX, for a few seconds, that will be the active VFO to respond (TX) to the incoming call. After those seconds, if no activity, the default TX is selected back. But then there comes the "Crossband" (omg...), where you can manually select which VFO want to be ALWAYS FIXED to respond.

In "T VFO" menu, "MAIN" should be renamed to something more like "AUTO" (if Dual Watch is active), or "SAME AS RX" (if inactive), while the other two should be renamed to "VFO A/UPPER VFO" and "VFO B/LOWER VFO". 

Doesn't matter how well we arrange this, I think this will be always be confusing as f***. Only bet I would say is safe is to place each options together in the menu (right now they are in position 10 and 23), so you can easiliy fix any mess.

egzumer, 2023-10-06:
> and in case of RX, for a few seconds, that will be the active VFO to respond (TX) to the incoming call

But that is my point, it doesn't work like that anymore. It always transmits on the frequency where the triangle is.

OneOfEleven, 2023-10-06:
It's been making my brain hurt from the start lol, hence why I've struggled to get the radio to TX on the correct VFO.

OneOfEleven, 2023-10-06:
Crossband is meant to TX on the opposite VFO to what your receiver is set too - for that's easy to understand.

The dual watch, the receiver can pause on either VFO (where ever their is a signal, but I think the TX should always be on the 'main' channel (solid triangle), otherwise your just never sure where you'll be transmitting.

The QS way is to TX on the frequency the RX has currently stopped on.

OneOfEleven, 2023-10-06:
I suppose if the RX pauses on a channel/opposite VFO, then maybe that's where the TX should TX, unless crossband is on, in which case it will transmit on the opposite VFO ... aarrgghhh

ea7hjb, 2023-10-06:
Ok, I rechecked everything again, and it doesn't work like the original QS firmware, but I think if it is not bad at all. Forget the Crossband (since that thing only purpose is to force in which VFO to transmit).

Right now, if you enable Dual Watch ("DUALRX" option 10), it will ALWAYS transmit to the selected VFO. No auto switch. So if 1o11 is gonna leave that thing that way, perhaps it would be clearer to rename menu options to: "OFF", "ON + TX VFO A" and "ON + TX VFO B"???

Then, now lets talk about the Crossband. If the Dual Watch is not going to do Auto temporary switching, then it can be DIRECTLY REMOVED, as it has no sense, because you are already selecting the TX VFO in the "DUALRX" option.

egzumer, 2023-10-06:
I don't know why they called what they implemented crossband. Repeaters are crossbands.

OneOfEleven, 2023-10-06:
mmm I know, it's really cross-vfo

egzumer, 2023-10-06:
I'm confused. Are we talking about 1 or 2 options in menu. The first is "T VFO"

egzumer, 2023-10-06:
Like it is right now it doesn't matter what I choose in the "T VFO" it will always transmit on the frequency with the solid triangle. So what changes between MAIN and UPPER/LOWER?

OneOfEleven, 2023-10-06:
T-VFO (QS call it WX) is the crossband thing, you set a VFO to TX on.

But then there is Dual-RX (QS call it TDR).

OneOfEleven, 2023-10-06:
I will investigate

ea7hjb, 2023-10-06:
It is confusing, pretty much. But 1o11 firmware works different than originals. And that is not a bad thing.

I will try again to explain how I see it's working in each firmware, but my english is not very good:

- Original firmware: if DW is active, both channels are monitored, when RX is detected, the TX is temporary auto switched to that VFO. If you do nothing, the TX goes back to the main VFO selected. If Crossband is active, you can have DW but you will override the auto TX selection and you will ALWAYS do TX where you want.
- 1o11 firmware: if DW if active, your monitor both VFO, but the TX will always occur on the active VFO. No auto switching to the last active VFO. Crossband? Is there, but now it's completely pointless. Can be removed unless 1o11 implements the auto TX switch on Dual VFO.

I am pretty sure of what I am saying. But what is best? who knows. Let the master decide.

OneOfEleven, 2023-10-06:
ok, the solid black triangle is/should always be the TX VFO - I think.

If you set just T-VFO (cross band without dual-rx), you transmit on the black triangle VFO, and RX on the other VFO (thin outline triangle).

egzumer, 2023-10-06:
What you saying is correct, but this modification doesn't make sense, lets not call it crossband, there is nothing crossband about it. The main point is to have option to respond to the last RX.

egzumer, 2023-10-06:
...if you choose

OneOfEleven, 2023-10-06:
lol

egzumer, 2023-10-06:
The original implementation didn't work either :) if I remember correctly. It always switched for couple seconds to the RX VFO

ea7hjb, 2023-10-06:
The only usefull option of "T VFO" right know would be, perhaps, to do TX on the other VFO without monitoring it.

egzumer, 2023-10-06:
It could be, but still I think responding to the last TX is useful function.

OneOfEleven, 2023-10-06:
Yes, it's for doing any TX-RX offset/split you want.

egzumer, 2023-10-06:
I might be on two frequencies with two groups of people, and respond to last call.

OneOfEleven, 2023-10-06:
The radio in my car does cross-band repeater mode, you can set RX on UHF say, and TX on VHF. Then you set your walkie talkies to use that config using the T-VFO option.

egzumer, 2023-10-06:
But the offset and direction does exactly that in QS

OneOfEleven, 2023-10-06:
> I might be on two frequencies with two groups of people, and migh want to respond to last call.

Then I guess that would be taken care of by the Dual-RX option ? (leave T-VFO off/main)

ea7hjb, 2023-10-06:
Would it be possible to merge both options in the same menu (I guess would be a PITA)?. If you don't plan to implement the auto TX switch on the DW (which I think is no big deal), perhaps it would be better to have a single menu with the following options:

VFO:
VFO A TX+RX
VFO B TX+RX
VFO A TX+RX/B RX
VFO A RX/B TX+RX
VFO A TX/B RX
VFO B RX/A TX

Oh my god. This is going to be a real mess. GJ Quansheng, lol

egzumer, 2023-10-06:
> Then I guess that would be taken care of by the Dual-RX option ? (leave T-VFO off/main)

It did work like that originally, not anymore :)

OneOfEleven, 2023-10-06:
> But the offset and direction does exactly that in QSO's instead.

It does - as well. but radio hams also like to do with with two separate VFO's instead

OneOfEleven, 2023-10-06:
> > Then I guess that would be taken care of by the Dual-RX option ? (leave T-VFO off/main)
> 
> It did work like that originally, not anymore :)

That will be my fault lol. Should I just go with what QS do, copy them ?  .. we have to power to do anything we want now though. We just need to know what we want.

egzumer, 2023-10-06:
Well I think it just shouldn't be called what QS called it. Other than that it was good I think. Could it be made better? Probably :)

ea7hjb, 2023-10-06:
Another option is to leave is working as the original QS method, with auto TX switching un DW, and rename menu to clearer strings:

Dual Watch (DUAL RX): OFF, PRIO(rity) VFO A, PRIO VFO B
Crossband (CROSS VFO): OFF, TX VFO A, TX VFO B

bricky149, 2023-10-06:
I've been investigating this as a means to build the dual-PTT functionality on and it looks possible. I've been struggling with button input so I haven't got it working properly, hence my branch isn't representative of what I've been doing.

I think I'll put the brakes on until this feature is worked out. I appreciate your efforts.

OneOfEleven, 2023-10-06:
We can make it better, we can rename it all to anything we like, we can even frequency hop (another day) !

egzumer, 2023-10-06:
I think I made my point I'll leave that for you to decide, I have to go to sleep. I'm away this weekend. Goodbye.

OneOfEleven, 2023-10-06:
Sleep sounds good. me too

RoadRangerX, 2023-10-16:
Rather than open another issue - I'd like it if the delay in "Dual Rx Respond" mode was adjustable, "a couple seconds" is a bit too short for me. :-)

OneOfEleven, 2023-10-17:
> Rather than open another issue - I'd like it if the delay in "Dual Rx Respond" mode was adjustable, "a couple seconds" is a bit too short for me. :-)

There is an S HOLD menu option for that.

closed 2023-10-17 by OneOfEleven