OneOfEleven/uv-k5-firmware-custom #135: F+* (scan) funcion gone.

ea7hjb, 2023-10-06:
Hello,

In some build (between e042b2 and fd3df27) the F+* (Scan) function, used for finding the CTCSS/DCS on a given frequency is no more. It always starts a normal frequency forward search.

However, it is important to note (I tried to record I video, but I should flash an old firmware again) that when that function was accesible, after exiting, it always recorded a different frequency in the VFO. Example: you are scanning the subtone of 446,00625 and found (or not) 67,0Hz. After pressing Menu ("SAVE?") and then Exit button, the VFO could read something like 430.000.

Hope it's easy to fix. Thanks in advance for your infinite patience.

OneOfEleven, 2023-10-07:
I know exactly why that is, I'll revert that. I disabled the code scan on F-key then */scan while I was looking for a bug (which I'm still looking for).

OneOfEleven, 2023-10-07:
Although the K5 user manual is difficult to understand at times due to the strange button descriptions/wordings, the F-plus-4 key was for CTCSS/DCS scanning ?

zhaoparrot, 2023-10-07:
QS default is :F-plus-* key for CTCSS/DCS scanning on current frequency , F-plus-4 for nearby TXradio frequency and CTCSS/DCS scanning

Romashca, 2023-10-07:
not really, F+4 is a frequency capture with or without a subtone. F+* is the definition of a subtone at the desired frequency (channel)

OneOfEleven, 2023-10-07:
oh I see.

So F+4 is suppose to do a small RF frequency scan (wonder how far to scan) for a very strong signal, then stops on it if found, then does a CTCSS/DCS scan on that signal ?

Romashca, 2023-10-07:
something like this

ea7hjb, 2023-10-07:
On build f93f365 the Frequency Hunter (F+4) seems broken. Doesn't find anything.

The Tone Hunter (F+*) displays a different frequency. Example: I am in VFO 446,00625, but it displays in the screen 446.01250 (+6.25kHz), but finds the tone. Weird, lol. The, it shows "'*' to save" but that does nothing?. If you do the F+* scan in memory, it will scan the frequency of the NEXT memory, so it looks like it wants to start scanning again, and manages to do a jump before entering the hunting function.

ea7hjb, 2023-10-07:
> oh I see.
> 
> So F+4 is suppose to do a small RF frequency scan (wonder how far to scan) for a very strong signal, then stops on it if found, then does a CTCSS/DCS scan on that signal ?

-40dB search, so must be very close indeed.

OneOfEleven, 2023-10-07:
Yes I've come across the worse bit of coding I've seen in some time, trying to make head nor tail of it is just not possible, so I'm in the process of fixing it to make it understandable, I first have to understand it lol.

Darn chinese !  they are fast at copying stuff, but design they are bad at.

rai4424, 2023-10-07:
You are right! F+4 for copy freq from other radio withen short distence and F+*  for scanning CTCSS/DCS for any receiving channel or freq.

ea7hjb, 2023-10-07:
> > P.D.: if you ask me, I would change the text from "FREQ scanning/CTC scanning" to "FREQ scanning/TONE scanning". >So is more symmetrical as both have 4 characters and, after all... it is searching for DCS tone too lol.
> 
> ah but .. DCS is not a tone, it's a digital LF code ;)

Well nomenclature... both are tone-coded squelch system, but who cares? doesn't really matter much ;)

Btw I don't know why you removed the rest of my comment, I guess you have already fixed that and the build is in queue to be publish, in which case I am really sorry for saying it again, but I double confirmed it, just with the best of my intentions, not to give you any false information.

spm81, 2023-10-07:
Bug is still there on OEFW-f93f365

Romashca, 2023-10-07:
by pressing F+*, scanning is enabled on the next channel in the channel list

OneOfEleven, 2023-10-07:
Been out in car. I'll find the problem.

ea7hjb, 2023-10-07:
> Been out in car. I'll find the problem.

No one is hurrying you. On the contrary, I think we all are thankful for your time. This is a hobby.

OneOfEleven, 2023-10-07:
Try it now, let me know how I goes. I made a mistake when I restored the F+* scanning this morning.

OneOfEleven, 2023-10-07:
I noticed in the original firmware it doesn't round the resulting RF frequency to the current step size, so wanted to fix that.

ea7hjb, 2023-10-07:
Working like a charm!. Looks like "*" key doesn't repeat, but the rest is working so great. Thank you!

OneOfEleven, 2023-10-07:
oh, * key should repeat the scan.

Does it not work for both F+4 and F+* scans ?

OneOfEleven, 2023-10-07:
Trying to find that-40dBm level you mentioned. Is it a value stated in firmware somewhere ?

ea7hjb, 2023-10-07:
Sorry, I couldn't responder earlier. I am now attending a wedding. Wish me
luck :(

Yes. * button didn't work neither on F+4 nor F+*.

The -40dB value was found long time ago by Tunas1337, if I remember well.
It's like the Scrambler Frequencies, but that thing deserves a long
conversation at another moment, and also the QS has a nasty, nasty bug that
perhaps someday it can be fixed.

OneOfEleven, 2023-10-07:
Just tested it. -6dBm (as far as the radio is concerned). Though the generator was saying -20dBm.

USBEprom, 2023-10-07:
Sorry to bother.
It works with CTS, but not with DCS.
The decoded DCS tone does not match the one issued.
Firmware 1o11-06c7d79
Thanks for you tireless work OneOfEleven!

OneOfEleven, 2023-10-07:
> Sorry to bother. It works with CTS, but not with DCS. The decoded DCS tone does not match the one issued. Firmware 1o11-06c7d79 Thanks for you tireless work OneOfEleven!

Nor does the original QS firmware unfortunately.

Well, it's only when trying to use the inverse DCS codes (ending with an 'I') that things don't work. The 'N; codes work.

USBEprom, 2023-10-08:
Hi OneOfEleven,
thanks for your kind reply.
As always you are totally right.
Sorry for the annoyance.

Be seeing you.
 
U.Sb

ea7hjb, 2023-10-08:
> oh, * key should repeat the scan.
> 
> Does it not work for both F+4 and F+* scans ?

Hello again. I am sorry but I have to withdraw my words. I would have sworn the * repeat button didn't work, but I re-checked again and it's working like a charm. Perhaps the keypad of my QS is starting to fail or (most likely) I'm becoming more stupid each day.

The F+4 Frequency Hunter finds the frequency with swiss precission, but if no tone is sent, it will stay forever in the "CODE searching". However I would say this is fine and you should not waste more time trying to fix this non-broken thing. User should have enought brain to understand that after a given time, if no code is found, it's because perhaps no code is being used? and press M key to stop and save!.

> > Sorry to bother. It works with CTS, but not with DCS. The decoded DCS tone does not match the one issued. Firmware 1o11-06c7d79 Thanks for you tireless work OneOfEleven!
> 
> Nor does the original QS firmware unfortunately.
> 
> Well, it's only when trying to use the inverse DCS codes (ending with an 'I') that things don't work. The 'N' codes work.
> 
> It seems to TX the correct code, but RX is broken for the inverse codes.

I would say that is not broken? Each DCS code has his own inverted code, so since we don't know the polarity of the transmitter the Quansheng is showing what is not inverted to him. I don't think this is working bad at all. Perhaps it's not showing "more information", but again, if you ask me, this is good enough. Here is a good and long reading for whom may be interested in understanding this weird DCS thing: https://web.archive.org/web/20210606183728/http://onfreq.com/syntorx/dcs.html

Having said that, thank again, one more time, for your time, effort, patience and good inclination towards perfection. This issue should be tagged as closed, I guess?

OneOfEleven, 2023-10-09:
No it's OK don't worry, I think I've got people worried about asking about adding a feature or not lol, don't want people to feel like that :)

I did see that it doesn't timeout on the code part of the scan if there is no code being transmitted, but you can restart the freq/code scan at anytime by pressing the **star** key, or **exit** to stop altogether, or as you say **menu** to save the frequency.

I shall add a timeout in the code part of the scan I think just so that the user is not left wondering what to do. I'll just put '**none**' on the code line under the frequency line to let them know no code was found.

OneOfEleven, 2023-10-09:
Just tried the wouxon handheld, that also doesn't detect a CDCSS inverted code, it displays an incorrect code. It detects normal CDCSS codes though (ending in 'N').

Seems it's a normal thing. I'll see if I can find a fix, another day though.

ea7hjb, 2023-10-09:
As I said, strictly speaking, I don't think we can say this is broken. It is just giving the right DCS code for the matching polarity. Showing the inverse DCS would be as simple as having a table with the opposite and printing it on screen as well. But that is something that can lead to many, many more ramifications (showing non-standard codes, the equivalent in number to PMR, etc...). Complications can be added as desired, but do you think they are worth your time? I would say better spend your time on something else, but it's your choice :P

And sorry for the offtopic (it's somehow related) but, is that KG-UV9K of yours capable of using scrambler? There is a nasty, nasty bug in the QS that, who knows, perhaps it even cannot be ever fixed, but if it can, for sure it will not be easy. Should I give you more information here?

Thanks, as always, for your time.

OneOfEleven, 2023-10-09:
What bug does the QS firmware have that you're referring to I wonder ?

No the KG-UV9 I have does not do voice inversion scrambling. The only reason I got the UV-K5's is for the scrambler ability, few radios have that ability these days. Some TYT radios have it though, the 9800 TYT radio in the car has it.

ea7hjb, 2023-10-09:
Well, OK, let's see how I explain this (my english is so so...): when the QS applies the inversion to the frequency you are listening, it does it to the whole signal. That includes the subaudible part that should not be processed. So what does that means? if CTCSS or even DCS is used, you will hear a continuos and very disturbing "beeeeeeeep", which is the CTCSS (or very fast "bip bip bip" if DCS). Turn off the CTCSS/DCS code in the transmitter et voilà, the "beeeep" is gone.

If you have two devices with scrambler, you can quickly check it. Right now I'm not home not fully squipped to test stuff and record a video. I don't know if I explained myself well or not, or if someone has observed the same thing as me.

So obviously the inversion should be applied after opening the squelch and removing the subtone from the signal. As I said, I don't even know if that is possible to get done or not. And if possible, for sure it will not be easy. Buy you can bet this doesn't happen on other deviced, name it Baofeng BF-888, Motorola CLP446, etc...

I will try to record a video as soon as I can.

closed 2023-10-10 by OneOfEleven