OneOfEleven/uv-k5-firmware-custom #136: Battery bar has nothing common with calibration steps made by amnemonic stript

pawol, 2023-10-07:
Battery bar has nothing common with calibration steps made by amnemonic stript

Here is my battery calibration:
Level 0: [1341] 5.20 V // Empty battery blinking
Level 1: [1805] 7.00 V // 1 battery bar if above this value
Level 2: [1921] 7.45 V // 2 battery bars if above this value
Level 3: [1960] 7.60 V // 3 battery bars if above this value, also value used to calculate adc to volt
Level 4: [2032] 7.88 V // 4 battery bars if above this value
Level 5: [2300] 8.92 V // overwritten by radio to 2300 anyway

even when voltage on screen is 7.35V still battery is full

egzumer, 2023-10-09:
This is the same as #139

pawol, 2023-10-10:
Seems works ok on commit 6660d00

OneOfEleven, 2023-10-10:
ah good , egzumer fixed that :)

closed 2023-10-10 by OneOfEleven