OneOfEleven/uv-k5-firmware-custom #137: [OT] Possible improvement: PAGER function.

USBEprom, 2023-10-07:
Sorry to bother, just a thought.

By removing the timing for the decoded DTMF string displayed on the display (the decoded DTMF string is displayed for an indefinite time until a subsequent decoding is successfully completed), it would be possible to obtain a simple PAGER function.

Doing the same thing with DTMF Decoding Response, it would be possible to obtain an advanced PAGER function.

Doing the same thing with DTMF Contact List, it would allow the sending of preconfigured PAGER messages up 8 characters max, like the Yaesu VX-3 Messages function (up 16 characters max), without losing the original DTMF Contact List functionality.

OneOfEleven, 2023-10-07:
The DTMF decoder just isn't that good to be honest, it detects characters that weren't sent all the time. That's why I use a timeout on it so to remove the many erroneous detections.

Much better off using the FSK modem in the radio.

OneOfEleven, 2023-10-08:
The radios do have a basic pager function in them already, although not with any message sending capability.

Receiving radio ..
* Set "**D DCD**" to **ON**
* Set "**D RSP**" to **RING** (or **BOTH**)
* Set "**D HOLD**" to time you want on-screen msg to remain

Transmitting radio ..
* Press the '*****' key to start the DTMF entry
* Enter the receiving radios "**ANI ID**" number, ie "**102**"
* Press the **PTT** to send it

The receiving radio should start to ring (beeping) and also display a 'called' message for however long "**D HOLD**" is set too.

It doesn't do message sending, but it does do basic paging.

USBEprom, 2023-10-08:
Hi OneOfEleven,
thanks for your kind reply.
I agree, the DTMF decoder detects characters that were not sent all the time and so it would be much better to use the FSK modem on the radio.
However, it may also be enough to have any decoded DTMF string, not necessarily agreed, which acts as a call warning if the operator is far from the radio when the message is received.
The operation can be improved by programming a specific tone to limit false positives, so that when the user sees a decoded DTMF string on the display, he reasonably knows that someone has called him.
Basically the Simple Pager/Pocket Beep function of Kenwood, Icom, Alinco, Yaesu and so on, works with that scheme.
The user who sends the message cannot be sure that it has been received by the recipient user and this latter can not be sure that the possible call warning is not the result of a false positive.

> The radios do have a basic pager function in them already, although not with any message sending capability.
> 
> Receiving radio ..
> 
>     * Set "**D DCD**" to **ON** .. enables DTMF decoding
> 
>     * Set "**D RSP**" to **RING** (or **BOTH**) .. enable beeping/ringing when DTMF call received
> 
>     * Set "**D HOLD**" .. sets the 'call from' message on-screen time

Yes, I know, that is exactly what I already do.
The limit of this is in the 60 maximum seconds that **D HOLD** allows at the call warning to remain visible on the display.
The use of **D RSP** makes it similar to the operation of the Enhanced Pager function provided by Yaesu.

> So it doesn't do message sending (at the moment), but it does do basic paging.

Actually, as I wrote, already now your firmware, but also the original one from Quansheng, allows the sending of messages.
The limits are in the 8 maximum characters for the text and 60 seconds maximum for their viewing on the display. 
Of course, as happens for example with the Yaesu VX-3, transceivers must all be programmed with the same messages.
A further limit is that it is not possible to write messages on the fly nor from the scratch, as it could be done with very old Kenwood, ICOM, Yaesu, Standard and so on (normally up  6 characters maximum), but there is the advantage that the messages cannot be decoded by third parties who do not know their programming code.

Be seeing you.
 
U.Sb

OneOfEleven, 2023-10-08:
The "**D HOLD**" menu setting does have a "**STAY ON SCREEN**" option. So you can set it to remain in screen until you next press a key.

USBEprom, 2023-10-08:
Hi OneOfEleven,
thanks, you are really great and very kind, the best!
Thanks a lot!

Be seeing you.
 
U.Sb

USBEprom, 2023-10-09:
Hi OneOfEleven,
as reference:

**D HOLD** set to **STAY ON SCREEN** works as expected for **DTMF Decoding Response**, while timing for the **decoded DTMF string** is not affected, it is always 5 seconds regardless **D HOLD** setting (**5 seconds** - **STAY ON SCREEN**).

Text "**CALL FRM:**" into **DTMF Decoding Response** prevents use of callsigns/messages 8 characters long, because the whole text exceeds the width of the display and so the last character start a new line generating artifacts.

Thanks a lot for your tireless and priceless work!

Be seeing you.
 
U.Sb

OneOfEleven, 2023-10-10:
Seems OK in this firmware.

I guess whatever QS is doing is also inverting the sub-audible up into the HF audio range.

USBEprom, 2023-10-11:
@OneOfEleven 

Hi OneOfEleven,
first, good to see you back.
I was sure you would have managed to fix things.

[image]

Sorry for the bad picture.
Text is "**CALL FRM : UV-5K(8)**"
Since the problem with CALLSIGNS/MESSAGES 8 characters long is only cosmetic and it can be easily arranged, for instance by removing colon from the text "**CALL FRM:**" to make it become "**CALL FRM**", thus saving a character on the total shown on the single line of the display and the possible modification is within the reach of all while it is building the firmware from themselves, in my opinion the issue can be closed.
Thanks a lot for your tireless support and great patience.

Be seeing you.
 
U.Sb

OneOfEleven, 2023-10-11:
No it's OK, I leave issues open so that I don't forget to sort them out, that's why I left yours open. I was going to get to it, just haven't got there yet.

USBEprom, 2023-10-16:
Hi OneOfEleven,
thanks for fixing the matter, you are really very kind.
Now it is gone.
Thanks a lot for your priceless support and great patience.

Be seeing you.
 
U.Sb

OneOfEleven, 2023-10-16:
You're welcome

closed 2023-10-16 by OneOfEleven