OneOfEleven/uv-k5-firmware-custom #138: Suggestion

lohtse, 2023-10-07:
Could I suggest you close the issues section for a while so that you can focus on yourself  and when you have time fixes the Bugs and issues you want too..

Alot of people seem to forget you have a life and are doing your best etc..

Just a suggestion to allow you to focus on what you feel is important  and yourself.

Thank you for what you have done and are doing..

OneOfEleven, 2023-10-07:
Yes I might well do for a while. I'm OK the moment, have done virtually nothing today with it, spending time with family.

lohtse, 2023-10-08:
> Wait... I'm supposed to have a life??

Shhhhhhhhhhhhhhhhhhhhhhhhus and build/ mod the damn firmware NOW.......

OneOfEleven, 2023-10-08:
No, you do as you're told, and when you're told !

bricky149, 2023-10-08:
Dropping by to say thank you for your work in taking the K5 to a new level. I had no experience of developing for an embedded platform until this $20 radio came along and fostered a community that I enjoy being part of. It was thanks in part to you that I started to tinker with radios as a hobby, got three already with a fourth on the way.

You are appreciated. Look after yourself and be well. 73

OneOfEleven, 2023-10-08:
I started developing on embedded systems around 1980 after my n-year apprenticeship was done when I moved into the design dept. It's a nice part of electronics, it empowers you to be able to create projects that go with your own ideas rather than someone else's or some companies idea.

These K5/K6's are so cheap at the moment, the day-to-day price of them goes up and down somewhat on aliexpress, one day they are less than £9 each (inc postage), the next they can be £18. Have no idea at all how they make them so cheaply, the workers must be paid next to nothing :(

bricky149, 2023-10-08:
Chances are the units are made en masse and people just stick their brand name on it. It wouldn't surprise me if there were over 100k in existence right now. I saw easily the most expensive K5 where I am in the UK, £50!

OneOfEleven, 2023-10-08:
I bought these two on aliexpress, they were here within 6 days. one was £13, the other was £10'is I think.

zhaoparrot, 2023-10-08:
> I started developing on embedded systems around 1980 after my n-year apprenticeship was done when I moved into the design dept. It's a nice part of electronics, it empowers you to be able to create projects that go with your own ideas rather than someone else's or some companies idea.
> 
> These K5/K6's are so cheap at the moment, the day-to-day price of them goes up and down somewhat on aliexpress, one day they are less than £9 each (inc postage), the next they can be £18. Have no idea at all how they make them so cheaply, the workers must be paid next to nothing :(
> 
> If you want to play more with the cortex CPU's, then buy some little STM32F103 boards or such like, maybe STM32F40x boards that have FPU's, the ST boards with in-built programming/debugging interface are really nice, the windows dev software lets you program and debug and single step through your code so easily.

video of QS K5 manufacture U may be interested↓
 https://b23.tv/r3rtLo1

ea7hjb, 2023-10-09:
I have another suggestion: why don't we leave open a single "issue" calles "Suggestions" in the Issues Section, where people can publish their ideas and, of course, OneOfEleven will evaluate if to attend those requests or pass, and leave the rest of the Issues Section only to post only real issues/bugs?.

I say this without any acrimony, and only with the best of the intentions of having the GitHub as organized as possible, thus avoiding having multiple issues open when in reality, they are not.

lohtse, 2023-10-09:
> I have another suggestion: why don't we leave open a single "issue" calles "Suggestions" in the Issues Section, where people can publish their ideas and, of course, OneOfEleven will evaluate if to [...]

Not actually a bad idea...

OneOfEleven, 2023-10-09:
Can do if you like, I don't mind either way to be honest. Pitty github didn't have a section for non-issues really.

I use the issues section to keep me reminded of thing to do though, if an issue gets closed before I've got around to it I'll forget all about it .. comes with age ;)

egzumer, 2023-10-09:
> Pitty github didn't have a section for non-issues really.

Actually there is a "discussions" section that can be turned on
`repo settings -> general -> features -> discussions`

wrcrooks, 2023-10-24:
+1 for a discussions section.

I'd like to get a discussion going on whether or not this radio has enough Non-Volatile Memory to implement DMR or M17 (by shamelessly stealing/duplicating the work that's being done in the OpenRTX codebase)

OneOfEleven, 2023-10-24:
The radio can never do DMR as it has no AMBI codec chip, and the little CPU has no access to the audio paths (TX or TX), plus it's grossly underpowered and flash/mem challenged for anything above basic control of the radio.

wrcrooks, 2023-10-24:
Well put, thank you sir!

closed 2023-10-28 by OneOfEleven