OneOfEleven/uv-k5-firmware-custom #140: Request for adding spectrum analyzer here.

rai4424, 2023-10-08:
Needs Spectrum analyzer here.
 Reasons:
1-Egzumer file is not updated.
2- Display tunrs blank during Tx.
3-Battery pesentage is  droping as   88%- 60%-30%-15%-06% at once and display starts glitching. 4-Screen was totally white two times on 80% Battery. I have nubers recordings but cant post here. 4-This issue was not in versions below V0.5. 
5-Mr. Egzumer does not respond on issue timely.
So its my humble request to add spectrum analyzer in your file.

lohtse, 2023-10-08:
Firstly it has been noted it is on the cards to add.. so leave it there

secondly I use egzumers daily and NEVER had the TX issue

thirdly   due to you supposed issue I have never had the battery issue

finally. devs ahev REAL LIFES AND ARE ALLOWED TO TAKE TIME OFF... SO be damn patientr and stop bloody pestering them...

rai4424, 2023-10-08:
I have video proofs of this issue.
There was no any issue after installing old version.

lohtse, 2023-10-08:
> I have video proofs of this issue. There was no any issue after installing old version.

Well I have custom dailys and NO ISSUE 

NOW STOP pestering the devs   and just friggin wait till they have time to respond  or just use an OLD verwsion that works for you..

Either way stop being a PITA

rai4424, 2023-10-08:
@lohtse You are taking it personl.
I have requested to OneOfEleven plz.

lohtse, 2023-10-08:
> @lohtse You are taking it personl. I have requested to OneOfEleven plz.

no I am just sick of people such as you pestering and when you don't get a response when you want it you pester further...  Its folks like you that get the issues section closed down  and devs leaving the k5 project... 

Just stop  simple

bricky149, 2023-10-08:
IIRC the dev started to integrate the spectrum analyser and may be seen in app/spectrum.c.

As they work on this in their spare time, self-care is important to help keep the hobby fun. I would ask not to request features and to make an attempt to troubleshoot issues you find yourself so that maybe it can help others.

rai4424, 2023-10-08:
Noted please
Thanks

closed 2023-10-08 by rai4424