OneOfEleven/uv-k5-firmware-custom #142: Disable TX completely feauture

kirishima-p, 2023-10-09:
TX function is illegal in depending on the country, 
I would like to have the ability to completely disable the TX function in all band.

OneOfEleven, 2023-10-09:
There is a TX enable/disable option in the normally hidden menu, you can also set some country/TX setting there too, although there are limited options.

To access the hidden menu items to change the TX options and if you want to completely disable the TX ..

* Turn the radio off
* Press and hold down the PTT and the upper side button at the same time
* Turn the radio back on whilst holding those two buttons down
* You should now have the extra/hidden menu settings to play with

Wonder what country you're in ?  I could add an option to match your part of the world to prevent the radio TX'ing where it's not allowed. The radio being able to TX where it shouldn't can get people into serious trouble and/or taken in for questioning in some parts of the world.

lohtse, 2023-10-09:
> There is a TX enable/disable option in the normally hidden menu, you can also set some country/TX setting there too, although there are limited options.
>
> To access the hidden menu items to change [...]

Probably means without a licence which is common throught the world....   So a generic disable everywhere could be useful for those without a licence etc

pawol, 2023-10-09:
Wrong thinking. Nobody will give you a fine for TX feature in a radio.

lohtse, 2023-10-09:
> Wrong thinking. Nobody will give you a fine for possessing TX feature in a radio. Without a licence you always transmit on your risk.

yes they will and DO in the UK and whats more you can even go to prison! Same with many countries if caught!

In the UK, it is illegal to use a two-way radio without an appropriate license. The penalties for doing so can vary depending on the circumstances of the offense, but can include:

Fines: If caught using a two-way radio without a license, you may be fined up to £5,000 or face imprisonment for up to 6 months.
Confiscation of equipment: Your two-way radio may be confiscated by the authorities if used without a license.
Prosecution: If found guilty of using a two-way radio without a license, you may face prosecution which can result in a criminal record.
Interference with emergency services: The use of unlicensed two-way radios can interfere with emergency services, such as the police or ambulance service. If this interference results in harm to a person or property, the offender may face a more severe penalty.
It is important to note that there are some exceptions to the license requirement, such as the use of certain types of radios in specific circumstances, or for use by licensed operators from other countries who are visiting the UK.

To avoid penalties and ensure compliance with the law, anyone who wishes to use a two-way radio in the UK should obtain the appropriate license and follow the rules and regulations set out by Ofcom, the UK communications regulator.

kirishima-p, 2023-10-09:
Thank you for reply.
I'm live in Japan,
In my country illegal for possession of unlicensed/uncertified transceiver with ready to tx'ing. (on any freqencies)
It's very complicated to get permission, so I modified to make it completely receive-only.

lohtse, 2023-10-09:
> Thank you for reply. I'm live in Japan, In my country illegal for possession of unlicensed/uncertified transceiver with ready to tx'ing. (on any freqencies) It's very complicated to get permission, so I modify to make it completely receive-only.

That sucks buddy... lets hope 1of11 or someone else can help.... in the mean time there is away to do it on stock fws

https://whosmatt.github.io/uvmod/?hidden

https://uvmod.valek.net.ru/?hidden       (google will translate and you have al fws too)

downside is poor AM  but at least its a start for you

OneOfEleven, 2023-10-09:
This firmware lets you completely disable the TX if you want, the hidden menu item "**TX EN**" lets you do it.

lohtse, 2023-10-09:
> This firmware lets you completely disable the TX if you want, the hidden menu item "**TX EN**" lets you do it.

Shows you how little I explorer the menus properly...lol

Great news for the op!

OneOfEleven, 2023-10-09:
Even in the USA where they maintain the illusion about how free they say they are they have rules about what to do with you if caught.

closed 2023-10-09 by OneOfEleven