OneOfEleven/uv-k5-firmware-custom #145: TX stuck and Step

hssbo, 2023-10-09:
I am using the most recent version... I activated the scan list and pressed PTT... then the radio got stuck in the Tx and I had to turn off the radio. The Scan I symbol remains on the display even when scanning list II Please include 5kHz step! Thank you for your excellent work!

OneOfEleven, 2023-10-09:
Hopefully fixed now.

hssbo, 2023-10-09:
Your work is **fantastic.** I will always be available to assist you with tests. Thank you very much!

OneOfEleven, 2023-10-10:
One bug I MUST be fixed is I've broken the power save function - again.

When the radio goes into power save mode the RX dies.

hssbo, 2023-10-10:
This seems like a serious problem! When you make a change to fix this, I can run a test and report the results. Thank you for your work!

OneOfEleven, 2023-10-10:
Found and fixed the power save bug, my fault as per normal ;)

closed 2023-10-10 by OneOfEleven