OneOfEleven/uv-k5-firmware-custom #150: Minro compile issue

lohtse, 2023-10-10:
if Beeps enabled it produces this error

OW_CHARGE_LEVEL -DENABLE_REVERSE_BAT_SYMBOL -DENABLE_CODE_SCAN_TIMEOUT -DENABLE_FREQ_CODE_SCAN_TIMEOUT -DENABLE_AM_FIX -DENABLE_SQUELCH_MORE_SENSITIVE -DENABLE_FASTER_CHANNEL_SCAN -DENABLE_RSSI_BAR -DENABLE_SHOW_TX_TIMEOUT -DENABLE_COPY_CHAN_TO_VFO  -I D:/TWatch/uv-k5-firmware-custom/ -I D:/TWatch/uv-k5-firmware-custom//external/CMSIS_5/CMSIS/Core/Include/ -I D:/TWatch/uv-k5-firmware-custom//external/CMSIS_5/Device/ARM/ARMCM0/Include -c driver/uart.c -o driver/uart.o
arm-none-eabi-gcc  -Os -Wall -Werror -mcpu=cortex-m0 -std=c11 -MMD -flto=2 -Wextra -DPRINTF_INCLUDE_CONFIG_H -DGIT_HASH=\"6660d00\" -DENABLE_FMRADIO -DENABLE_UART -DENABLE_SMALL_BOLD -DENABLE_VOX -DENABLE_KEEP_MEM_NAME -DENABLE_WIDE_RX -DENABLE_1250HZ_STEP -DENABLE_TX_WHEN_AM -DENABLE_F_CAL_MENU -DENABLE_CTCSS_TAIL_PHASE_SHIFT -DENABLE_BOOT_BEEPS -DENABLE_SHOW_CHARGE_LEVEL -DENABLE_REVERSE_BAT_SYMBOL -DENABLE_CODE_SCAN_TIMEOUT -DENABLE_FREQ_CODE_SCAN_TIMEOUT -DENABLE_AM_FIX -DENABLE_SQUELCH_MORE_SENSITIVE -DENABLE_FASTER_CHANNEL_SCAN -DENABLE_RSSI_BAR -DENABLE_SHOW_TX_TIMEOUT -DENABLE_COPY_CHAN_TO_VFO  -I D:/TWatch/uv-k5-firmware-custom/ -I D:/TWatch/uv-k5-firmware-custom//external/CMSIS_5/CMSIS/Core/Include/ -I D:/TWatch/uv-k5-firmware-custom//external/CMSIS_5/Device/ARM/ARMCM0/Include -c app/action.c -o app/action.o
arm-none-eabi-gcc  -Os -Wall -Werror -mcpu=cortex-m0 -std=c11 -MMD -flto=2 -Wextra -DPRINTF_INCLUDE_CONFIG_H -DGIT_HASH=\"6660d00\" -DENABLE_FMRADIO -DENABLE_UART -DENABLE_SMALL_BOLD -DENABLE_VOX -DENABLE_KEEP_MEM_NAME -DENABLE_WIDE_RX -DENABLE_1250HZ_STEP -DENABLE_TX_WHEN_AM -DENABLE_F_CAL_MENU -DENABLE_CTCSS_TAIL_PHASE_SHIFT -DENABLE_BOOT_BEEPS -DENABLE_SHOW_CHARGE_LEVEL -DENABLE_REVERSE_BAT_SYMBOL -DENABLE_CODE_SCAN_TIMEOUT -DENABLE_FREQ_CODE_SCAN_TIMEOUT -DENABLE_AM_FIX -DENABLE_SQUELCH_MORE_SENSITIVE -DENABLE_FASTER_CHANNEL_SCAN -DENABLE_RSSI_BAR -DENABLE_SHOW_TX_TIMEOUT -DENABLE_COPY_CHAN_TO_VFO  -I D:/TWatch/uv-k5-firmware-custom/ -I D:/TWatch/uv-k5-firmware-custom//external/CMSIS_5/CMSIS/Core/Include/ -I D:/TWatch/uv-k5-firmware-custom//external/CMSIS_5/Device/ARM/ARMCM0/Include -c app/app.c -o app/app.o
app/app.c: In function 'APP_TimeSlice10ms':
app/app.c:1490:67: error: expected ')' before 'AUDIO_PlayBeep'
 1490 |   if (g_boot_counter_10ms > 0 && ((g_boot_counter_10ms % 25) == 0)
      |      ~                                                            ^
      |                                                                   )
 1491 |    AUDIO_PlayBeep(BEEP_880HZ_40MS_OPTIONAL);
      |    ~~~~~~~~~~~~~~
app/app.c:1799:1: error: expected expression before '}' token
 1799 | }
      | ^
make: *** [app/app.o] Error 1
Press any key to continue . . .

Nothing major  so can be ignored for the most part

OneOfEleven, 2023-10-10:
It's OK, just chilling at moment having a browse through bits of the code, playing with wordings n stuff.

Seems the boot beeps isn't working anyway, I broke it.

I thought I spotted a bug earlier with it ignoring a key press again. Let me know if you spot it happenings.

lohtse, 2023-10-10:
My default is always alarm disabled  but now am getting this 

/uart.o
arm-none-eabi-gcc  -Os -Wall -Werror -mcpu=cortex-m0 -std=c11 -MMD -flto=2 -Wextra -DPRINTF_INCLUDE_CONFIG_H -DGIT_HASH=\"d212d44\" -DENABLE_FMRADIO -DENABLE_UART -DENABLE_SMALL_BOLD -DENABLE_VOX -DENABLE_TX1750 -DENABLE_KEEP_MEM_NAME -DENABLE_WIDE_RX -DENABLE_1250HZ_STEP -DENABLE_TX_WHEN_AM -DENABLE_F_CAL_MENU -DENABLE_CTCSS_TAIL_PHASE_SHIFT -DENABLE_BOOT_BEEPS -DENABLE_SHOW_CHARGE_LEVEL -DENABLE_REVERSE_BAT_SYMBOL -DENABLE_CODE_SCAN_TIMEOUT -DENABLE_FREQ_CODE_SCAN_TIMEOUT -DENABLE_AM_FIX -DENABLE_SQUELCH_MORE_SENSITIVE -DENABLE_FASTER_CHANNEL_SCAN -DENABLE_RSSI_BAR -DENABLE_SHOW_TX_TIMEOUT -DENABLE_COPY_CHAN_TO_VFO  -I D:/TWatch/uv-k5-firmware-custom/ -I D:/TWatch/uv-k5-firmware-custom//external/CMSIS_5/CMSIS/Core/Include/ -I D:/TWatch/uv-k5-firmware-custom//external/CMSIS_5/Device/ARM/ARMCM0/Include -c app/action.c -o app/action.o
app/action.c: In function 'ACTION_AlarmOr1750':
app/action.c:275:44: error: unused parameter 'b1750' [-Werror=unused-parameter]
  275 |  static void ACTION_AlarmOr1750(const bool b1750)
      |                                 ~~~~~~~~~~~^~~~~
cc1.exe: all warnings being treated as errors
make: *** [app/action.o] Error 1

OneOfEleven, 2023-10-10:
Hopefully fixed, let me know

lohtse, 2023-10-10:
You're a star... all fixed

closed 2023-10-10 by lohtse