OneOfEleven/uv-k5-firmware-custom #151: FMradio and lock issue

lohtse, 2023-10-10:
user reported that when using FMradio and the device locks they cannot unlock....  I can confirm the issue  and it only happens when FMradio is in use
[image]

lohtse, 2023-10-10:
I can also confirm this issue did not exist before at least the 4th Oct(checked previous FWs)

OneOfEleven, 2023-10-10:
You're right !

Just looking now.

OneOfEleven, 2023-10-10:
I've fixed that.

The keys used in the FM radio are not good, I'm slowly going through the keyboard stuff there, such as using the 3/VFO/MR for swapping between channel and VFO mode instead of some other key (strange).