OneOfEleven/uv-k5-firmware-custom #153: Bad magic number when reading EEPROM

ANTodorov, 2023-10-10:
Hello,

I am trying to read the EEPROM using [k5prog](https://github.com/sq5bpf/k5prog).

At random moments during the reading of the eeprom the communication is failing with error "bad magic number". Also the station hangs on "uart config comms" until I remove the KW-type cable.

Please find attached the logs of two failing attempts and a successful dump with OEFW flashed on the same device. It is interesting that the returned "bad magic" is same readable string `ps w` in any of the failing reads.

Model: UV-K5
FW: 1o11-a552b04
Comm cable: Baofeng branded CH341 UART serial
k5prog cmdline: `./k5prog -p /dev/ttyUSB0 -vvvv -r -f uv-k5-eeprom-oefw.bin 2>&1 | tee uv-k5-eeprom-oefw.log`

[uv-k5-eeprom1.log](https://github.com/OneOfEleven/uv-k5-firmware-custom/files/12857934/uv-k5-eeprom1.log)
[uv-k5-eeprom2.log](https://github.com/OneOfEleven/uv-k5-firmware-custom/files/12857935/uv-k5-eeprom2.log)
[uv-k5-eeprom-oefw.log](https://github.com/OneOfEleven/uv-k5-firmware-custom/files/12858141/uv-k5-eeprom-oefw.log)

ANTodorov, 2023-10-10:
I am confirming the k2prog tool successfully dumped the EEPROM. Works with the QS's CPS software too.

An other, minor, issue noted - the welcome voice on power on speaks both English and Chinese welcome phrases.

Should I close this issue and open another one or you'll handled this?

OneOfEleven, 2023-10-10:
Speaks both languages, like one after another ?

ANTodorov, 2023-10-10:
Yep. Starts with English and ends with Chinese. The end of the Chinese
is instantly followed by speaking the channel name in english (it is
booting in memory mode, not vfo)
It is like "Whelcome-tianchinsomething-channel-x". Only the welcome
message. all other voices sounds OK (in english)

ANTodorov, 2023-10-10:
Switched to VFO mode. On Power on it is like welcome-two-two-chin(different chiunese)-Frequency-mode

OneOfEleven, 2023-10-10:
oops, have you got a backup of all your channels etc ?

ANTodorov, 2023-10-10:
Please don't worry - nothing lost. it is entirely experimenting device.
Here is my eeprom.  The transceiver is powering on in VFO on A channel and Mem on B channel:
[uv-k5-eeprom6.bin.gz](https://github.com/OneOfEleven/uv-k5-firmware-custom/files/12860726/uv-k5-eeprom6.bin.gz)

OneOfEleven, 2023-10-10:
oh good. Just glad you have a backup of your settings.

No it's OK I'm just looking for the problem.

ANTodorov, 2023-10-10:
Just let me know if there is something I could help with.
Here is the first "successful" eeprom read. The transceiver was booting in Mem on both A and B channels (well with the funny welcome. I was eager to test the eeprom reading and then looked at the other things.
[uv-k5-eeprom4.bin.gz](https://github.com/OneOfEleven/uv-k5-firmware-custom/files/12860818/uv-k5-eeprom4.bin.gz)

ANTodorov, 2023-10-11:
Hello,
The eprom read via k5prog works with latest build too.

I'll close this isasue and open a new one with the welcome sound as it is irrelevant to this one

closed 2023-10-11 by ANTodorov