OneOfEleven/uv-k5-firmware-custom #154: Problem with chirp

Aquila161, 2023-10-10:
Hi, I have to report a problem, I loaded the channels with chirp, and the initial message got corrupted, I put the original firmware back and did a total reset, but the radio no longer loads the channels even with the original program, and the message remains always like this:
[image]

Aquila161, 2023-10-10:
With the PSCPS program it asks me for a password

OneOfEleven, 2023-10-10:
Mine too :(

My poor little K5 (and yours too by the sounds of it) have suffered a blow tonite, one which I'm going to take a long look at tomorrow to see what I can do to get them back.

My fault :(

Romashca, 2023-10-11:
I have the same problem. the program requires a password , but the chirp does not flash . trouble

OneOfEleven, 2023-10-11:
I've got my radio back !! :) .. although I just seen your message.

I've been investigating to find out exactly what's gone wrong, so that I can create a proper solution (and fix) to this mishap .. the eeprom data got wrongly over-written/corrupted due to a bug I created last night.

The AES key in my radio got overwritten/corrupted (among other things), once I noticed that I wrote a short bit of code (a few mins ago) to wipe the AES key back to default (0xfffff ..).

Don't want this happening again, so a recovery bit of code in the firmware is going to be added.

Best thing anyone can do is to backup the ENTIRE 8192 (0x2000) bytes of their radios EEPROM, that way they will have a backup they can use to recover their radio with. I did this from the start using the k5prog-win I wrote, I'll add info and instructions shortly.

I know it's only a £10 (or so) radio (price changes daily), no great loss, but still don't want anyone to loose theirs (or me mine), seems a huge waste to do so.

OneOfEleven, 2023-10-11:
But, their is still a bug in my firmware in the UART section that uploads/downloads the radios setup, I want to trace and fix that before I upload anymore commits.

wrcrooks, 2023-10-11:
Just for your awareness, I have CHIRP working on my UV-K6, running 1o11-769b933. Don't know if that helps or not.

wrcrooks, 2023-10-12:
@Romashca But there isn't a version of the CPS for Linux, right? CHIRP has the best support by far IMO.

IU4FIR, 2023-10-13:
I don't know if it can be useful, but I would like to point out that loading data using ORIGINAL CPS QS does not work correctly.
The software confirms that the loading took place correctly but the changes to the memories are not loaded on the radio. 
I'm using 1o11-bc416c7
Thank you so much for the great work you do!

OneOfEleven, 2023-10-13:
All the channels end up with incorrect frequencies ?

I know chip gets things wrong when you use frequencies outside of the original QS firmware range. That's a problem with chirp though (and the QS PC software), the firmware can do nothing about that, the firmware only saves to memory what chirp and the QS software actually silently passes to it.

IU4FIR, 2023-10-13:
No, the memories remain the original ones both in terms of frequency and name, no incorrect data, however all the memories I use are within the standard range of the radio.
I did a test by reloading the original fw and then loading the modified cps, the changes were inserted into the radio. 
Subsequently I reloaded your fw, it turned out that the memories had remained updated without any errors.
A comment about the new fw, the feedback is very positive as AM modulation, really clean. 
Thanks again and have a great weekend!

OneOfEleven, 2023-10-13:
My fault, try it now, let me know.

closed 2023-10-13 by OneOfEleven

IU4FIR, 2023-10-13:
Installed the df41bfa version, it works! tnx