OneOfEleven/uv-k5-firmware-custom #156: Good to see You Back

lohtse, 2023-10-11:
Good to see you back

And don't ever worry  99.9% would not blame you for issues when flashing as we all make proper back ups etc.. And we accept errors can happen when building FW's

as to the compiling there is a minor issue but I am sure you will resolve it once you have sort things out fully

As always THANK YOU so much for the DAMN hard work you put in on a £15 radio...  What you have done has been quite frankly,  amazing  and the vast majority are as grateful as hell too you for it.

regards
Lohtse

egzumer, 2023-10-11:
Yes, thank you for your hard work, errors happen, don't worry about that. The software is provided as is, no money exchanged.

Although I would recommend using main, devel, and maybe feature branches.
`main` - somewhat stable version, completed features
`devel` - work in progress
feature branch - separate branch for certain feature in progress.

This is not a github thing so you don't have to get friends with it :)

OneOfEleven, 2023-10-11:
Thank you.

I hid the repo last night to try and keep the radio losses to a minimum, I needed to find the problem.

Solution to mishap I now have :)

lohtse, 2023-10-11:
> Thank you.
> 
> I hid the repo last night to try and keep the radio losses to a minimum, I needed to find the problem.
> 
> Solution to mishap I now have :)

Take your time to fix them... It is not like there are not other FW's too chose from...

And my apologies for the behaviour of a certain individual... He kicked up a fuss again on Telegram..

OneOfEleven, 2023-10-11:
You've been buying them all up then lol   Anyway, you're welcome !

I've been taking it slow today, been doing a bit on the FM radio to make it nicer to use.

OneOfEleven, 2023-10-11:
Had a look on youtube for K5 videos today, just seeing if anything new around, it's scary when you see your work being used else where around the world. All totally due to DualTachyon's amazing work though !

OneOfEleven, 2023-10-11:
> Although I would recommend using main, devel, and maybe feature branches. `main` - somewhat stable version, completed features `devel` - work in progress feature branch - separate branch for certain feature in progress.
> 
> This is not a github thing so you don't have to get friends with it :)

lol

Do I somehow setup a 2nd branch to achieve that I wonder ?  and then upload/commit the the **devel**, then on occasion, somehow copy that to the main branch if all is well ?

LZ2DMV, 2023-10-11:
> Do I somehow setup a 2nd branch to achieve that I wonder ? and then upload/commit to the **devel** twig rather than the main one, then on occasion, somehow copy that to the main branch if all is well ?

Yes, that is the general idea. You just create a new branch, say 'dev', commit the changes part of the development process there, and when you consider it to be stable enough, you do the "copy" to 'main'. In Git, the "copy" has two forms - merge and rebase, each one of them with its advantages and limitations. [Here](https://www.atlassian.com/git/tutorials/merging-vs-rebasing)  is some nicely steuctured information.

And [here](https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-and-deleting-branches-within-your-repository) is how you create a branch using the GitHub web interface. After you sync your repository with whatever Git client you are using later, the new branch must be there. If your client is a CLI one, after syncing with the remote repository (`git pull`), you can do `git branch -a` in the repository folder to list all your branches and do `git checkout [branch_name]` to switch to a branch.

Thank you very much for everything that you are doing for the community!

egzumer, 2023-10-11:
To add to that, I recommend using Tortoise Git as a graphical interface for git, it makes things much easier for me.

OneOfEleven, 2023-10-11:
I tried the github desktop app/software last night, with undesired consequences, it kindly shared some personal details without first asking me :(

I use tortoisegit to upload commits to github - at the moment.

egzumer, 2023-10-11:
You don't need github desktop I didn't even know it exists, everything you can do in the tortoise git, then you simply push the commits, branches and so on to the github repository. 

Simplest instruction: To merge `dev` to `main` you switch to `main` and in the commit log click right on `dev` and hit `Merge to main`. 

If you're not used to it there will be some challenges I'm sure. You have to play with it a little bit :)

OneOfEleven, 2023-10-11:
Yes I did a full uninstall of github desktop after that.

If you see any new branches pop up, just ignore them for now, it'll only be me trying things out.

egzumer, 2023-10-11:
I see only main and revert, they don't appear in github until you push them, so you can play with it all you want locally.

ea7hjb, 2023-10-12:
As many did, I would to thank you for your time and effort.

We all understand (and the one who doesn't, too bad for him/her) we are doing what we shouldn't. And it's very fun. And comes with risk (that can be minimized if people do their homework, aka backups). And all of this is thank to you. Period.

OneOfEleven, 2023-10-13:
Now have sublime installed, looks nice. I'll see how it goes.

wrcrooks, 2023-10-14:
Might I throw VSCode into the ring as a contender? I'm a huge fan of their Source Control integration. It's very useful/powerful, even for someone like me who doesn't quite understand how Git works :joy:

OneOfEleven, 2023-10-14:
I still haven't managed to configure VSCode to work with the project yet :(

wrcrooks, 2023-10-14:
I'm working with Linux but the concept should be the same. You do a "git clone" to a working directory on your computer and you open the "uv-k5-firmware-custom" directory with VSCode, then VSCode sees the ".git" folder and does the rest. You then find all your Git actions in the left vertical menu under "Source Control"

Hope that helps! Keep up the great work!

closed 2023-10-28 by OneOfEleven