OneOfEleven/uv-k5-firmware-custom #158: Dual watch not start on radio boot

reppad, 2023-10-11:
Hi,

When I turn on the radio, the dual watch doesn't start (it's activated in the menu), the radio stays stuck indefinitely with the `><` logo until I do certain actions.
It's not just the display, the unselected VFO is not receiving communications.

What activates the dual watch
- press Menu
- press a numbered key
- press an arrow keys in channel mode
- receive a communication on the activated VFO

What doesn't activate dual watch
- exit key (takes the radio out of power save mode, but does not activate dual watch)
- the PTT key
- press an arrow keys in frequency mode

I observed this behavior on my last 2 tests (commits 774e34bf and 5942582d).

Thanks for your work, I'm glad to see your repo online again !

OneOfEleven, 2023-10-11:
Yes I was playing with the voice stuff for the first time, didn't quite understand it's various settings.

Hopefully fixed now.

reppad, 2023-10-11:
Yes, I did a quick test and it works.
I don't know if it's from the last commit but the FM also seems to work better than yesterday.
Thanks!

closed 2023-10-11 by reppad

OneOfEleven, 2023-10-11:
I'm just fixing the key presses when in FM radio mode.

reppad, 2023-10-11:
I must have been in worse reception conditions when I tried to use FM yesterday, it's very difficult to measure reception quality "by hand", there are so many parameters to take into account...