OneOfEleven/uv-k5-firmware-custom #164: compile issue

Romashca, 2023-10-12:
ui/main.c:88:4: error: implicit declaration of function 'UI_PrintStringSmallBold'; did you mean 'UI_PrintStringSmall'? [-Werror=implicit-function-declaration]
   88 |    UI_PrintStringSmallBold(s, 2, 0, line);  // issue UI_PrintStringSmallBold  //UI_PrintStringSmall(s, 2, 0, line);
      |    ^~~~~~~~~~~~~~~~~~~~~~~
      |    UI_PrintStringSmall
cc1.exe: all warnings being treated as errors
make: *** [ui/main.o] Ошибка 1

Romashca, 2023-10-12:
app/dtmf.c:418:29: error: 'g_dtmf_call_state' undeclared (first use in this function); did you mean 'gDTMF_CallState'?
  418 |                         if (g_dtmf_call_state != DTMF_CALL_STATE_NONE           ||
      |                             ^~~~~~~~~~~~~~~~~
      |                             gDTMF_CallState
app/dtmf.c:418:29: note: each undeclared identifier is reported only once for each function it appears in
app/dtmf.c:419:29: error: 'g_current_vfo' undeclared (first use in this function); did you mean 'gCurrentVfo'?
  419 |                             g_current_vfo->DTMF_ptt_id_tx_mode == PTT_ID_APOLLO ||
      |                             ^~~~~~~~~~~~~
      |                             gCurrentVfo
app/dtmf.c:423:33: error: 'g_dtmf_reply_state' undeclared (first use in this function); did you mean 'gDTMF_ReplyState'?
  423 |                                 g_dtmf_reply_state = DTMF_REPLY_NONE;
      |                                 ^~~~~~~~~~~~~~~~~~
      |                                 gDTMF_ReplyState
make: *** [Makefile:371: app/dtmf.o] Error 1 

compile from docker

OneOfEleven, 2023-10-12:
I don't even know what docker is. need to delete those two files.

Romashca, 2023-10-12:
arm-none-eabi-size firmware
   text    data     bss     dec     hex filename
  57268      28    2816   60112    ead0 firmware 

docker 
FROM archlinux:latest
RUN pacman -Syyu base-devel --noconfirm
RUN pacman -Syyu arm-none-eabi-gcc --noconfirm
RUN pacman -Syyu arm-none-eabi-newlib --noconfirm
RUN pacman -Syyu git --noconfirm
RUN pacman -Syyu python-pip --noconfirm
RUN pacman -Syyu python-crcmod --noconfirm
WORKDIR /app
COPY . .

#RUN make && cp firmware* compiled-firmware/

OneOfEleven, 2023-10-12:
???

Romashca, 2023-10-12:
> ui/main.c:88:4: error: implicit declaration of function 'UI_PrintStringSmallBold'; did you mean 'UI_PrintStringSmall'? [-Werror=implicit-function-declaration] 88 | UI_PrintStringSmallBold(s, 2, 0, [...]

UI_PrintStringSmall(s, 2, 0, line);  // issue   edit :UI_PrintStringSmall(s, 2, 0, line);
thats work

OneOfEleven, 2023-10-12:
Just had a look, you uploaded that change yourself.

Romashca, 2023-10-12:
I wana help

OneOfEleven, 2023-10-12:
Yes sure, no problem :)

Romashca, 2023-10-12:
Thanks for you work

closed 2023-10-12 by Romashca

OneOfEleven, 2023-10-12:
try it now, just committed (myself ;) )