OneOfEleven/uv-k5-firmware-custom #165: AIRCOPY

OneOfEleven, 2023-10-12:
Just been looking closer at the the way QS have implemented this AIRCOPY feature is very flawed :(

It reads/writes directly from/too the eeprom, but as it's transferring the data from one radio to the other, their is no ability to cope with any errors at all, which means any missed packets (of which happens every time I try it) means that particular part of the RX radio's eeprom remains un-updated :(

The implementation needs to be changed to include ACK packets from the RX radio so that the TX radio can resend the missed packets, otherwise the RX radios eeprom is going to be left in a mixed up state containing both the TX radios eeprom data and the RX radios previous eeprom data .. eeek

So be warned, if at the end of using AIRCOPY you see the RX radio show any errors at all (the 'E' number), you MUST restart/redo the complete transfer again, and keep doing it until it does a complete error free run !

I've been writing raw packet handling systems most of my life (including AX.25), I can completely solve the problem by fixing the implementation using ACK packets from the RX radio, but then the AIRCOPY feature would no longer be compatible with radios that use the original QS firmware.

I spose I could add an option to the AIRCOPY screen to use the current flawed QS implementation, or a fixed new one.

zhaoparrot, 2023-10-12:
they should not release product like this . very irresponsible.
the part of eeprom it was writing will never broke the radio evenif the data was mixed i presume?

OneOfEleven, 2023-10-12:
The area of eeprom that is copied between radios is the entire configuration, but not including the calbration data area, but the transfer does include the AES code and the password the user might be using. I found out what a corrupted AES code can do the other night ;)

OneOfEleven, 2023-10-12:
I can add protection to the RX aircopy code to protect those two values though (AES code and password).

OneOfEleven, 2023-10-12:
The radios also transmit the data to the other radio at full TX power.

I've tested with setting the TX output set to a bare minimum (mW's), it's all that's needed, unless you want to copy to a radio a few miles away ?

zhaoparrot, 2023-10-12:
> The radios also transmit the data to the other radio at full TX power.
> 
> I've tested with setting the TX output set to a bare minimum (mW's), it's all that's needed, unless you want to copy to a radio a few miles away ?

i think in some case, nearby transmit with max power may actually generate more error than minimum power.

LZ2DMV, 2023-10-14:
> I've been writing raw packet handling systems most of my life (including AX.25), I can completely solve the problem by fixing the implementation using ACK packets from the RX radio, but then the AIRCOPY feature would no longer be compatible with radios that use the original QS firmware.

Maybe add the ACK extension, but with a Makefile option (and make it 0 by default), so people who flash your firmware on all of their radios can benefit from it?

OneOfEleven, 2023-10-14:
Yes I was actually thinking about doing that, just that, I haven't - yet, keep getting side tracked, or chilling.

OneOfEleven, 2023-10-14:
I think I'll make it so that it continues as per normal QS way, but add code that listens out for any possible replies/acks from the RX'ing radio, if it gets any then act on them, ie resend any corrupted/lost packets so that the RX'ing radio can fill in the gaps in the eeprom.

LZ2DMV, 2023-10-14:
Yiu mean if it gets at least one ACK back, it knows it is talking to another OneOfEleven radio, so it enters ACK mode and keeps expecting ACKs for every frame sent from now on?

OneOfEleven, 2023-10-14:
Not really, I just simply get the TX'ing radio to resend that one missing packet if it receives an ack from the RX'ing radio.

The way the aircopy transfer works is the TX'ing radio sends it's eeprom contents in small blocks/packets, 120 of them to be exact.

closed 2023-10-16 by OneOfEleven