OneOfEleven/uv-k5-firmware-custom #167: wide FM (standard radio service) cannot be received by communication receiver

pawol, 2023-10-13:
wide FM (standard radio service) cannot be received by communication receiver.

This issue is known since the beginning of QS. 
For instance when we set 104,90MHz (local wide stereo FM radio service in my area) then we are not able to hear it clearly.
Demodulation is awfull. 
Question is why. Set of wide band and FM should be enough to hear it

OneOfEleven, 2023-10-13:
Not sure what you're asking ?

FM broadcast deviation is like 50kHz, narrow band comms deviation is 5kHz or less.

pawol, 2023-10-13:
Aaa... ok. If standard radio service has 50kHz of deviation then it makes sense on 25kHz wide band of communication chanel

OneOfEleven, 2023-10-13:
If you want to listen to FM broadcast station, you need to use the wide band FM radio.

pawol, 2023-10-13:
ok, issue cleared

closed 2023-10-13 by pawol

OneOfEleven, 2023-10-13:
Some countries have already stopped using the VHF FM broadcast band for broadcast radio, it's now used for others things.