OneOfEleven/uv-k5-firmware-custom #169: SQUELCH TAIL NOSE

rai4424, 2023-10-14:
Noticing squelch tail noise after installing cistom FW.
Before, was not present in K5 to K5 and baofeng to K5. Thanks in advance.

wrcrooks, 2023-10-14:
What are your settings for the following?
- ROGER
- PTT ID
- STE

rai4424, 2023-10-14:
> What are your settings for the following?
> 
> * ROGER
> * PTT ID 
> * STE.  

ROGER Off
 PTT ID  Off
 STE   On
I have checked with 03 talkies(k5+k5+Baofeng A58) with this setting now. Problem is partially solved.  When K5 with custom Fw Txing, other two make SQ tail noise.

I will be thankful if u clarify for me its funtion.
( I want to close this this issue soon to avoid disterbence for respectful programer)
Thanks

OneOfEleven, 2023-10-14:
Squelch tail noise is normal for FM, actually unusual not to hear it.

The only way to mute it is to use say CTCSS or CDCSS which has the option to TX a mute tone/code at the end of your TX in order to tell the RX'ing radio to mute its audio early, and so muting the normal FM squelch tail.

I think that's what you're referring to ? .. not long woke up.

OneOfEleven, 2023-10-14:
STE is the option that tells your TX to send the CTCSS/CDCSS mute tone/code (but only if you're using CTCSS/CDCSS).

OneOfEleven, 2023-10-14:
ROGER and PTT ID are for different uses, mainly to annoy radio hams.

OneOfEleven, 2023-10-14:
CTCSS/CDCSS happens in the background at the same time as everything else, you don't hear it as it's normally filtered out in the RX'ing radio before it reaches the speaker.

It's like when elephants and whales talk to each other, they use LF audio that hour hears don't respond to.

closed 2023-10-14 by rai4424