OneOfEleven/uv-k5-firmware-custom #176: Feature Request: AGC for Microphone

cloudwindy, 2023-10-15:
Is it possible to automatically adjust mic gain based on sound level?

cloudwindy, 2023-10-15:
Since I haven't come up with a practical solution, let's make this a discussion.

OneOfEleven, 2023-10-15:
Yes it's easy to do. I'll add it at some point, busy though now with other things for a while.

RoadRangerX, 2023-10-16:
I was thinking that an option to enable the Tx compression without the Rx expansion (Compandor On) would do this? I was just about to put that as a feature request when I saw this issue :-)

OneOfEleven, 2023-10-17:
You can certainly enable the TX compander if you want, it will give more TX audio mic gain to those that want/need it.

bricky149, 2023-10-24:
I believe this is possible without the compander. It's part of #218, AGC can control both MIC (Tx) and IF (Rx) inputs.

RoadRangerX, 2023-10-28:
Does the Tx Compandor actually work? With the mic gain all the way up I still have low modulation and turning the Tx Compandor on doesn't seen to do anything? Any possibility of more gain in the firmware?
I'm using egzumer's build 0.14.1

OneOfEleven, 2023-10-28:
Yes the compander works.

Have no idea why you have low TX audio. Just make sure you use WIDE mode with receiving radios that are set to wide mode.

RoadRangerX, 2023-10-28:
This graph from audio recorded off my local repeater shows another operator first, me with Compandor on and max mic gain, and with Compandor off. Yes, the Tx Compandor does work - if I blow this up I can see it doing a bit of gain and limiting peaks.
[PoorModulation](https://github.com/OneOfEleven/uv-k5-firmware-custom/assets/24575272/7d9ef0c8-0bcf-464b-9d32-38ee0f530df9)

OneOfEleven, 2023-10-31:
Is the latest firmware any better ?

RoadRangerX, 2023-11-01:
> Is the latest firmware any better ?

I'm using egzumer's builds. I see he dropped release v0.15 yesterday, hopefully he brought in whatever change you made in your code?

Might take me a couple days - my 1TB SSD totally died this morning and I need to redo whatever I set up in the last 2 or three months which includes the UK-K5 software :-(

Thanks! Ron