OneOfEleven/uv-k5-firmware-custom #181: 70cm band sensitivity

xawen, 2023-10-16:
There seems to be a difference in rx quality and sensitivity on the 70cm band between this firmware and the stock OEFW.  

I do not have proper test equipment for this, but I have tested as thoroughly and consistently as possible with what I have.  If there are any other comparisons that you would find helpful (without test equipment), I'm happy to do them.

Test setup:
UV-K5 with Nagoya NA-701
Local repeater on 442.3 Mhz

The repeater was active with a conversation between two fixed base stations.  During this conversation, I flashed the radio between several different firmwares and observed the results.  Observations were made only when base station #1 was transmitting (to eliminate any variability between operators).  The same radio and antenna were used.  The radio was on battery power during testing.  Each firmware was flashed and tested twice, results did not change between flashes.  The entire test ran for about 25 minutes.

Each firmware was tested in two locations.  The radio was put in the same position during each test.
1) 2nd floor desk near a window - good reception of the repeater
2) Basement table - borderline reception of the repeater

Unfortunately, I did not record the build versions of each firmware (I can re-test if this would be useful).  But each was built from a fresh pull of their repository on 10/15.  All firmware tested had the recent RSSI fix, but I can't confirm they all measure RSSI the same way.  The RSSI number is included for reference.  Main point of comparison was the audio output.  

Results
OneOfEleven
   Desk
      - RSSI average of -82
      - Audio clear with constant static
      - Squelch remained open
   Basement
      - RSSI average of -101
      - Heavy static, voices audible at high volume
      - Squelch intermittently open and closed

Stock OEFW
   Desk
      - No RSSI available
      - Audio clear with infrequent static
      - Squelch remained open
   Basement
      - No RSSI available
      - Audio clear with constant static
      - Squelch remained open

Stock OEFW - RSSI mod (quick modification I made to display RSSI)
   Desk
      - RSSI average of -61
      - Audio clear with infrequent static
      - Squelch remained open
   Basement
      - RSSI average of -83
      - Audio clear with constant static
      - Squelch remained open
  
Fagci
   Desk
      - RSSI average of -62
      - Audio clear with infrequent static
      - Squelch remained open
   Basement
      - RSSI average of -83
      - Audio clear with constant static
      - Squelch remained open

OneOfEleven, 2023-10-16:
okey doke. I'll have a look with the signal generator.

I presume this is in FM mode (not AM) ?

The only thing I change from OEFW is the front end gains, and ONLY in AM mode,. FM mode uses all the same settings, so in theory their shouldn't be any difference, but I may have done something not quite right somewhere.

ea7hjb, 2023-10-16:
As far as I am aware, I think this may has been inherited from OEFW, as this has been observed on other OEFW based firmwares as well.

OneOfEleven, 2023-10-16:
I do make the squelch more sensitive, so it's up to the user to step up the squelch setting in the menu if it's a bit too sensitive for them, which is exactly what the squelch level setting is for.
Those that will ONLY ever use squelch level 1 when there are 9 or 10 to choose from is not making good use of the available settings the user has to play with.

I do update the squelch changes I make from time to time if it seems they're not at their best though. So reports are welcomed (don't want to scare people off)

xawen, 2023-10-16:
> 
> I presume this is in FM mode (not AM) ?
> 

Yeah, this was all in FM.  It's been mentioned a few times in Telegram, so I'm pretty sure it's not just my hardware.  But no one had done any A/B testing to confirm.

rai4424, 2023-10-22:
Yes I have lost sensitivity. I have compare original  and custom Fw and found a difference in receiving week signals. SQUELCH was not useful in that situation.

bricky149, 2023-10-24:
The issue appears to be because AGC is never enabled. I have ported the fix from fagci's FW.

xawen, 2023-10-24:
> The issue appears to be because AGC is never enabled. I have ported the fix from fagci's FW.

I picked your changes and am testing.  Thanks for finding this.

xawen, 2023-10-25:
After the AGC fix, reception on the 70cm seems to be improved (on my radio at least).  The RSSI reported on the signal bar has not changed (and is still ~20 lower than reported from other firmwares).  But the audio clarity and the ability of the radio to keep the squelch open are both much better after the change.  I can no longer hear a difference when compared to the same radio running base OEFW or Fagci.  So it's either a good improvement, or a very effective placebo.

OneOfEleven, 2023-10-25:
I'm sure I'll put pay to that at some point ;) !

bricky149, 2023-10-25:
> After the AGC fix, reception on the 70cm seems to be improved (on my radio at least). The RSSI reported on the signal bar has not changed (and is still ~20 lower than reported from other firmwares). [...]

Some of the code is swapped between EnableAGC and DisableAGC so I'm unsure if you got the placebo or not. As I've said elsewhere, I'll need to wait until I'm back surrounded by air traffic before I can continue. AGC should be applicable to mic input so maybe I can try auto mic gain while I'm at it.

OneOfEleven, 2023-10-25:
It's because I went through my code and fixed the front end settings.

I've also disabled AGC.

OneOfEleven, 2023-10-25:
don't forget, '0' = AGC (auto), '1' = fixed (no AGC). in reg 0x7E

OneOfEleven, 2023-10-25:
also, you'll break the AM fix if you use anything but reg 0x13 for the front end gains. change that and you'll upset a lot of people.

bricky149, 2023-10-25:
> don't forget, '0' = AGC (auto), '1' = fixed (no AGC). in reg 0x7E

I had it backwards, my apologies. That would mean it's on by default, if the docs are to be believed.

> also, you'll break the AM fix if you use anything but reg 0x13 for the front end gains. change that and you'll upset a lot of people.

I understand. I was confident this morning thinking I'd stumbled on something but now I'm not so sure. Feeling the Dunning-Kruger effect for sure.

bricky149, 2023-10-25:
Many thanks for your patience, and to those who gave my thoughts the time of day (even if I was wrong).