OneOfEleven/uv-k5-firmware-custom #185: SSB

Rikku2000, 2023-10-17:
Can you please add option for SSB?

i see in code:
BK4819_AF_BASEBAND1 =  4u,  // SSB
BK4819_AF_BASEBAND2 =  5u,  // SSB

i think its LSB/USB or?

ct1esj, 2023-10-17:
That would be gold on blue

OneOfEleven, 2023-10-17:
I was thinking about it, but need to add a few more frequency step sizes along with adding SSB (RX only) to make SSB usable, something like 100Hz and 500Hz steps.

ct1esj, 2023-10-17:
Yes, of course, it makes perfect sense, but it would be great

Rikku2000, 2023-10-17:
That's easy on your firmware, just incress

Frequency.c also change in header:
const uint16_t STEP_FREQ_TABLE[10] = {50, 100, 125, 250, 500, 625, 1000, 1250, 2500, 833};

OneOfEleven, 2023-10-17:
Can't add it like that, that would create a mess with the stepping as it has to remain compactible with what the windows software uploads to radio radio, it would put **everyones** steps out of syn, the meaning of all the step settings would all be wrong.

Also their only one bit allocated by the windows software for setting mode to FM or AM, I'd need to add code to include another bit elsewhere in the eeprom to allow the extra modes.

Rikku2000, 2023-10-17:
Or you can do en separate #ifdef enable_ssb

OneOfEleven, 2023-10-17:
At the moment, out of flash space for adding anymore.

Rikku2000, 2023-10-17:
@OneOfEleven do you have a messager, maybe we can get in contact.

closed 2023-10-17 by OneOfEleven