OneOfEleven/uv-k5-firmware-custom #189: menu on 2e880a6

IU4FIR, 2023-10-18:
### reporting some bugs with no priority  :-))
The key beep in some menu steps does not work, it is not always the same function, scrolling up and down the beep become silent in a random way. 
At menu step 45 of 61 a "1" symbol appears at the top right, I don't think it means anything.-->picture
**Nothing needs to be resolved, I'm just reporting for your information only.**

Best 73
[image]

ct1esj, 2023-10-18:
Yep, here I noticed the same thing

OneOfEleven, 2023-10-18:
Thanks for letting me know.

I know about the random missing key beeps, been looking for the reason for it, I'll find it eventually.

The little '1' that appears is saying what the current sub menu number is, it also tells you in the CTCSS and CDCSS menus (if they are not set to 'off'). I'll make it so that the little number only shows up when you enter the sub menu.

OneOfEleven, 2023-10-18:
I found where the missing beeps were going, have brought them back - hopefully.

IU4FIR, 2023-10-18:
Great job, regarding the submenus I had thought of something similar but I hadn't analyzed it completely.
 I apologize for the incorrect reporting...

closed 2023-10-18 by IU4FIR