OneOfEleven/uv-k5-firmware-custom #194: Discussion: FSK Message Support

cloudwindy, 2023-10-20:
When looking at [aircopy.c](https://github.com/OneOfEleven/uv-k5-firmware-custom/blob/main/app/aircopy.c) I came up with this idea: can we use the same code for transmitting text messages?

OneOfEleven, 2023-10-20:
Yes, the 1200/2400 FSK modem can be used for various things, such as text msg, AX.25, paging, syncing rolling code voice inversion, anything at all where data transfer is useful.

Lar-Sen, 2023-10-20:
Would love to have an inbuilt POCSAG encoder/decoder. A simple Baycom-type modem would be splendid, too. A first attempt was done, as I see in "fsk-uart" branch of Tunas1337 https://github.com/Tunas1337/uv-k5-firmware/tree/fsk-uart but seems forgetten since a long time.

OneOfEleven, 2023-10-20:
Well that's something you would each have to implement yourselves because this version of the firmware is something I'm doing for my own needs/desires, I just happen to be sharing it as it is.

The FSK modem also does MSK (bypasses the 1200/1800/2400 Hz FSK), so feel free to develop a POCSAG util.

Lar-Sen, 2023-10-20:
I know about your hard work, and improving things while reading constant suggestions for new features is somewhat frustrating, for sure :)  Mass storage is very limited too. I guess even fully wiping aircopy function wouldn't leave enough room to implement POCSAG frame building and others...

Anyway POCSAG is 2-level direct FSK, 16K0F1D for most common version. FLEX uses 4-level. What should be feasible is a Baycom-type serial interface, with an option in menu to select FSK / AFSK, and let another computer-side software or app do the job.

Like many others here, I'm pretty ambitious but lack some skills :)  Keep up the good job!

OneOfEleven, 2023-10-20:
When the wants get a bit too much I just disable the issues github option for a short while, it stops the influx of emails I get so that I can do my own thing for a while without the bother, even if it is to just chill out and watch movies :)

cloudwindy, 2023-10-20:
@OneOfEleven Thank for your good job too. Also don't hesitate to let me know if there's anything I can help!

wrcrooks, 2023-10-20:
> Well that's something you would each have to implement yourselves because this version of the firmware is something I'm doing for my own needs/desires, I just happen to be sharing it as it is.

DualTachyon's repo points here for modded firmware so I feel that's where a lot of the (misguided) expectations are coming from. Would it be beneficial to have a more "public" repository with more contributors? I don't know if you much care for the PRs that I've submitted to add certain functionality but the bottom line is that you get to decide whether or not you want it in this repo. Your work is by far the best I've seen in terms of end user customization, through the use of the Makefile definitions.

OneOfEleven, 2023-10-20:
People are free to clone or fork or whatever the code if that's what they want.

This ones mine though.

closed 2023-10-20 by OneOfEleven