OneOfEleven/uv-k5-firmware-custom #197: Compilation error: unused variable 'smallest_char_spacing'

wrcrooks, 2023-10-20:
During compile:

```
ui/main.c: In function 'UI_DisplayMain':
ui/main.c:403:21: error: unused variable 'smallest_char_spacing' [-Werror=unused-variable]
  const unsigned int smallest_char_spacing = ARRAY_SIZE(g_font3x5[0]) + 1;
                     ^~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make: *** [Makefile:476: ui/main.o] Error 1
```

Solution: Commenting out line 403 in ui/main.c

OneOfEleven, 2023-10-20:
Presume you're using big fonts ?

wrcrooks, 2023-10-20:
Yes sir.
```
ENABLE_BIG_FREQ                  := 1
ENABLE_SMALL_BOLD                := 0
```

PieeerB, 2023-10-20:
+ channel to vfo copy dont work

OneOfEleven, 2023-10-20:
yes it does

wrcrooks, 2023-10-20:
@OneOfEleven Compiles now. Thanks!

closed 2023-10-20 by wrcrooks