OneOfEleven/uv-k5-firmware-custom #202: Radio enables various options at once when changing away from AM airband

bricky149, 2023-10-21:
I usually have the local ATIS on tap but I'm currently somewhere that doesn't get it. I changed the frequency to a local repeater on 430MHz and the radio decided to enable the compander, disable TX power, enable DTMF and introduce a lag that doesn't update the chosen frequency.

I'll look at it when I can, but I'm guessing there's a short-circuit in the logic somewhere that's flipping the wrong bits.

OneOfEleven, 2023-10-21:
I'm wondering if you have the 350MHz hidden option set to disable ? .. un-hide the hidden menu items by holding the PTT and upper side button, then go have a look at the 350MHz setting

there is a bug that I'm looking for that disables the 350MHz to 400MHz range when doing an aircopy.

bricky149, 2023-10-22:
I had 350 TX disabled but 350 EN enabled. Turning the TX option on seems to have resolved it but I don't understand why. I don't use aircopy so I disable it when compiling.

OneOfEleven, 2023-10-23:
I think I'm seeing something similar that's connected to this bug. Am on it.

closed 2023-10-23 by OneOfEleven