OneOfEleven/uv-k5-firmware-custom #203: PTT on external mic

BLUESALOON, 2023-10-21:
I noticed that when an external mic is connected the radio **will not tx using mic PTT** on a busy channel/active repeater.  
**BCL set to OFF.**
The PTT on the radio **will tx** the radio on a busy channel with external mic connected.
The PTT on external mic **will tx** radio on inactive channels.
I have tried multiple external mics, all with the same result.
Thank you very much for all your hard work
James   MM6WER

OneOfEleven, 2023-10-21:
There is something strange going on with the QS radios when it comes to using an external microphone. I'm told that if the original QS microphone that comes with some radios works OK, but when trying to use other mics like baofeng etc things get strange with the radio. I don't yet know why.

BLUESALOON, 2023-10-21:
Thanks for your reply

closed 2023-10-21 by BLUESALOON