OneOfEleven/uv-k5-firmware-custom #204: PTT ID Setting Does Not Save Across Reboots

wrcrooks, 2023-10-21:
To reproduce:
1. Power on radio
2. Switch to MR/Channel Mode
3. Open menu, navigate to PTT ID
4. Set to Apollo Quindar
5. Exit menu
6. Transmit to hear PTT ID
8. Power off radio
9. Power on radio
10. Transmit to hear no PTT ID

If I had to guess, `g_request_save_channel = 1` is not actually saving the PTT ID channel settings in the `APP_time_slice_500ms()` method?

I appreciate all of your work on this project!! (And sorry to add to your workload)

OneOfEleven, 2023-10-21:
PTT ID is a per channel setting (and per VFO), so if you set it while on a channel, you have to save the channel to get it to remember. That's how all the per channels are. I was auto saving the channel for some per channel settings, but got told to stop doing it and revert it back.

OneOfEleven, 2023-10-21:
It's the same with L/M/H TX power and various other settings, they are only remembered is you save the save after setting them.

wrcrooks, 2023-10-21:
You're a rockstar. Thanks mate!

closed 2023-10-21 by wrcrooks

OneOfEleven, 2023-10-21:
lol, the thought of being famous would finish me in an instant, I've always valued complete anonymity above most all things in life ;)

wrcrooks, 2023-10-21:
Well one of eleven rockstars then, that's close enough to anonymity right? :joy: