OneOfEleven/uv-k5-firmware-custom #205: Honestly speaking

rai4424, 2023-10-22:
OneofEleven's updated K5 is the best FW if following  problem have been solved and spectrum analyzer be added.
1- Poor reception on Uhf band, 462.00MHz. Reception is main work of a transceiver which can not be compromise.
2- Chanel name fonts are not bold.
3- Signal meter has disappeared.
I have to go to stock FW for communication with mates. I hope Oneofeleven will take serious action against these problems and we will able to use our favorite Firmware.
Thanks

bricky149, 2023-10-22:
1. Reception performance can vary based on weather, environment, etc. This is not strictly a firmware issue.
2. Suggestion, not an issue.
3. Cannot confirm. Works For Me(tm)
4. Cannot confirm. I've got two radios and I can't get them to transmit clicks.

The readme makes it clear this is provided as-is and is used at your own risk. Based on how you're presenting this, I'd suggest you stick to stock firmware.

DualTachyon, 2023-10-22:
Don't forget to request a refund if you're not happy with this firmware.

closed 2023-10-22 by rai4424