OneOfEleven/uv-k5-firmware-custom #207: small bold font

IU4FIR, 2023-10-22:
Maybe I'm missing something, but is there an option to use the bold character for the channel name, without using the compiler?
73?
[image]
[image]

OneOfEleven, 2023-10-22:
no

The reason for the compiler options is to remove code from the firmware. So disabling the bold font frees up flash bytes needed by something else.

You need to compile with yourself with your desired compiler options.

IU4FIR, 2023-10-22:
Ok, but the only way to modify the fw is by using the method indicated in your README.md file; it is not possible to use "VISUAL CODE" or other native Windows software?

closed 2023-10-22 by IU4FIR