OneOfEleven/uv-k5-firmware-custom #208: VHF Band II

pdrbsts, 2023-10-22:
Please change the FM broadcast lower frequency from 88 to 87.5 MHz.
https://en.wikipedia.org/wiki/Band_II

ea7hjb, 2023-10-22:
FM should be from 76 to 108, as some countries begin in 76 (Japan, 76-95, Brazil 76-108). So leaving 76-108 you are sure to always be right. It could be programmable, but I'm not sure it we should ask OneOfEleven to do that job. Having it wide open and letting the user use his/her brain should be the recommended choice.

OneOfEleven, 2023-10-22:
I can add a hidden menu option so the user can set the FM radio limits.

ea7hjb, 2023-10-22:
> I can add a hidden menu option so the user can set the FM radio lower limit being as it varies around the world.
> 
> The chip can do 68 to 108, but I doubt the RF front end filter covers the full range, but don't know without checking it on the signal generator.

It is not up to me do decide but, running so low of space, as you are well aware, is that necessary? If you want to make it programmble, would it not be better (in terms of space) to decide that during compile time in the Makefile?

OneOfEleven, 2023-10-22:
Either really.

I'll go the compile option route then, you're right, it won't use any extra flash bytes.

closed 2023-10-22 by OneOfEleven